marvinroger/async-mqtt-client #1: Is OnSubscribeCallback undocumented?

jpmens, 2016-05-11:
Glancing at the [example program](https://github.com/marvinroger/async-mqtt-client/blob/master/docs/1.-Getting-started.md), I don't see where a message for a subscription is received. Is it possible that `OnSubscribeCallback()` is undocumented or am I just simply blind? :-)

marvinroger, 2016-05-11:
* **onConnect**: when successfully connected (not only to the network, but when the connection is acknowledged from the broker)
* **onDisconnect**: when disconnected from the network (network down or bad credentials, etc...)
* **onSubscribeAck**: when the broker has acknowledged the subscription
* **onUnsubscribeAck**: when the broker has acknowledged the unsubscription
* **onPublish**: when the broker has sent a message to us. This is what you're looking for :)
* **onPublishAck**: when a QoS 1 or 2 message you've published is acknowledged by the broker

I know there is no documentation for this, this is in the TODO list. :)

marvinroger, 2016-05-14:
How is it going?

jpmens, 2016-05-14:
Haven't had the time to experiment yet, unfortunately.

jpmens, 2016-05-31:
#facepalm I thought you'd just pasted existing method docs there; after re-reading this three times I saw the hidden message _This is what you're looking for_. :-)

I am now testing.

jpmens, 2016-05-31:
@marvinroger I am very disappointed: I can't break it! 😆 

I have a test sketch publishing in `loop()` every 10ms, and I'm bombarding the same sketch with publishes:

```
test/lol count 75559 / nmess=389888
```

75k pubs from the device, 390k received by the device. This is very good work!

jpmens, 2016-05-31:
I do feel that the naming of these functions is confusing:

```
onPublish: when the broker has sent a message to us. This is what you're looking for :)
onPublishAck: when a QoS 1 or 2 message you've published is acknowledged by the broker
```

This is probably due to my experience with the Mosquitto C library, which has an `on_subscribe()` callback for messages *received*. So, from the point of view of the app on the device, I think `onSubscribe` makes more sense...

marvinroger, 2016-05-31:
Great news!

I am sorry, I am unable to find a reference to what you're saying. However [callbacks from libmosquitto](http://mosquitto.org/man/libmosquitto-3.html#idm46437505493968) are as follow:

* on_connect (eq: onConnect)
* on_disconnect (eq: onDisconnect)
* on_publish (eq: onPublishAck)
* on_message (eq: onPublish)
* on_subscribe (eq: onSubscribeAck)
* on_unsubscribe (eq: onUnsubscribeAck)

Messages received are sent to the `on_message` callback in libmosquitto. I would happily rename the methods to these ones. But `onSubscribe` means *when I'm subscribed* more than *when I receive a message*, don't you think?

jpmens, 2016-05-31:
Sorry: I of course _meant_ `on_message`.

If you ask me, it would be a good idea to use the Mosquitto function call names.

jpmens, 2016-05-31:
@marvinroger This is unbreakable; I give up! ;-)

Sketch has been running for 7.5 hours, and has published 2,444,038 messages in 10 ms intervals. Killing off the broker several times for minutes at a time (and then restarting it obviously) hasn't bothered your code in the slightest.

Fantastic.

marvinroger, 2016-06-02:
@jpmens kudos to @me-no-dev for the underlying [ESPAsyncTCP](https://github.com/me-no-dev/ESPAsyncTCP), he's the one who permitted this. I only implemented the MQTT client protocol. :)

marvinroger, 2016-06-03:
I guess the FullyFeatured example sketch is self-explanatory now. :)

jpmens, 2016-06-03:
I think so too ;-) FYI, 15 million messages in/out without a hiccup; a couple of reconnects, but that's fine.

Also FYI (even if half off-topic here) is 33 days of uptime on a Homie device. :-)

closed 2016-06-03 by jpmens