marvinroger/async-mqtt-client #2: example doesn't build.

matbor, 2016-05-19:
Using the example, https://github.com/marvinroger/async-mqtt-client/blob/master/docs/1.-Getting-started.md

I get the following error

```
Arduino: 1.6.8 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

Build options changed, rebuilding all
/var/folders/0s/sftdlnmx35scwybt6gn15s640000gp/T/arduino_modified_sketch_466625/sketch_may19a.ino: In function 'void onMqttDisconnect(AsyncMqttClientDisconnectReason)':
sketch_may19a:23: error: no matching function for call to 'HardwareSerial::println(AsyncMqttClientDisconnectReason&)'
   Serial.println(reason);
                        ^
/var/folders/0s/sftdlnmx35scwybt6gn15s640000gp/T/arduino_modified_sketch_466625/sketch_may19a.ino:23:24: note: candidates are:
In file included from /Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Stream.h:26:0,
                 from /Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/HardwareSerial.h:31,
                 from /Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Arduino.h:246,
                 from sketch/sketch_may19a.ino.cpp:1:
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:79:16: note: size_t Print::println(const __FlashStringHelper*)
         size_t println(const __FlashStringHelper *);
                ^
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:79:16: note:   no known conversion for argument 1 from 'AsyncMqttClientDisconnectReason' to 'const __FlashStringHelper*'
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:80:16: note: size_t Print::println(const String&)
         size_t println(const String &s);
                ^
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:80:16: note:   no known conversion for argument 1 from 'AsyncMqttClientDisconnectReason' to 'const String&'
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:81:16: note: size_t Print::println(const char*)
         size_t println(const char[]);
                ^
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:81:16: note:   no known conversion for argument 1 from 'AsyncMqttClientDisconnectReason' to 'const char*'
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:82:16: note: size_t Print::println(char)
         size_t println(char);
                ^
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:82:16: note:   no known conversion for argument 1 from 'AsyncMqttClientDisconnectReason' to 'char'
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:83:16: note: size_t Print::println(unsigned char, int)
         size_t println(unsigned char, int = DEC);
                ^
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:83:16: note:   no known conversion for argument 1 from 'AsyncMqttClientDisconnectReason' to 'unsigned char'
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:84:16: note: size_t Print::println(int, int)
         size_t println(int, int = DEC);
                ^
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:84:16: note:   no known conversion for argument 1 from 'AsyncMqttClientDisconnectReason' to 'int'
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:85:16: note: size_t Print::println(unsigned int, int)
         size_t println(unsigned int, int = DEC);
                ^
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:85:16: note:   no known conversion for argument 1 from 'AsyncMqttClientDisconnectReason' to 'unsigned int'
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:86:16: note: size_t Print::println(long int, int)
         size_t println(long, int = DEC);
                ^
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:86:16: note:   no known conversion for argument 1 from 'AsyncMqttClientDisconnectReason' to 'long int'
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:87:16: note: size_t Print::println(long unsigned int, int)
         size_t println(unsigned long, int = DEC);
                ^
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:87:16: note:   no known conversion for argument 1 from 'AsyncMqttClientDisconnectReason' to 'long unsigned int'
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:88:16: note: size_t Print::println(double, int)
         size_t println(double, int = 2);
                ^
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:88:16: note:   no known conversion for argument 1 from 'AsyncMqttClientDisconnectReason' to 'double'
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:89:16: note: size_t Print::println(const Printable&)
         size_t println(const Printable&);
                ^
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:89:16: note:   no known conversion for argument 1 from 'AsyncMqttClientDisconnectReason' to 'const Printable&'
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:90:16: note: size_t Print::println()
         size_t println(void);
                ^
/Users/matthewbordignon/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.h:90:16: note:   candidate expects 0 arguments, 1 provided
exit status 1
no matching function for call to 'HardwareSerial::println(AsyncMqttClientDisconnectReason&)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Obviously removing the ``onMqttDisconnect`` function it will build.

Can I also suggest you add the example to the source as well when you get time.

Matt.

marvinroger, 2016-05-19:
Yes, my bad, I've been changing quite a lot of things since I wrote the example. I'll fix and try to maintain the examples updated. :)

matbor, 2016-05-19:
Thx, thought as much, also your link in the docs;

"1a. For the Arduino IDE
Download the latest release"
https://github.com/marvinroger/async-mqtt-client/blob/master/docs/1.-Getting-started.md

is broken atm.

matbor, 2016-05-19:
```
Exception (28):
epc1=0x4022722c epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000002c depc=0x00000000

ctx: sys 
sp: 3ffffce0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe80:  00000000 00000000 0000001f 401050f1  
3ffffe90:  4020496d 3ffeeda8 0000013b 4010020c  
3ffffea0:  3ffef000 000000f8 3ffeeda8 40203b67  
3ffffeb0:  00000000 000000f8 3ffeeda8 40203bab  
3ffffec0:  3fffdab0 00000000 3ffeeda8 40203bcc  
3ffffed0:  402276dd 3fff01fc 00000001 40204896  
3ffffee0:  402276ba 00000000 3ffef0e0 40204a56  
3ffffef0:  40227957 00000000 3fff0a74 40204a73  
3fffff00:  40227482 3fff0a74 3ffe96e1 40227589  
3fffff10:  3ffe96fd 3ffe9719 00000000 40226f51  
3fffff20:  00000000 3ffeec14 002ceab9 40228938  
3fffff30:  00000000 00000000 3ffeec14 002ceab9  
3fffff40:  40228941 002d093b 3ffec850 40106da0  
3fffff50:  40228bb8 3ffe8bf4 002d093b 40228bb1  
3fffff60:  00000000 4021bca9 3ffedf98 3ffedfc0  
3fffff70:  018e8b91 60000600 3ffedfc0 018e29ec  
3fffff80:  4021bcee 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffedfd8 3fffdad0 3ffef0e0 402055bf  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Builds/uploads but crashes on trying to connect to broker.

Will wait until you have a working example to test.

Matt.

closed 2016-05-19 by marvinroger

marvinroger, 2016-05-19:
It should be good now.

matbor, 2016-05-19:
thx, will test tonight.

commit c01187c0 (2016-05-19): Fix #2 and #3
commit c79a83f0 (2016-05-19): Actually fix #2
commit ec60ac05 (2016-10-10): Implement multiple event handlers - fix #2