marvinroger/async-mqtt-client #4: Call ESP8266httpUpdate within on_mqtt_publish. Return connection refused

kenpeter, 2016-05-24:
Call ESP8266httpUpdate within on_mqtt_publish. ESP8266httpUpdate returns connection refused. i.e. not able to download firmware. If I call ESP8266httpUpdate within void setup(). It is able to download the firmware.

[image]

Full code is here.
https://github.com/kenpeter/mqtt_esp_http_update_sample/blob/master/phone.txt
https://github.com/kenpeter/mqtt_esp_http_update_sample/blob/master/device.txt

The idea is that the phone.txt fire a json command, then it triggers on_mqtt_publish in device.txt, then it calls ota_update()

marvinroger, 2016-05-24:
ESP8266httpUpdate won't work in the callback anyway, because ESP8266httpUpdate is a synchronous library. Try to set a flag (instead of your `ota_update()`), and check from within your main `loop()` if this flag is set, and if so call your `ota_update()`.

kenpeter, 2016-05-26:
Thank you. This makes sense.

closed 2016-06-02 by marvinroger