marvinroger/async-mqtt-client #5: Last Will

drejkus, 2016-06-04:
Hi, 
I just wanted to note (for anyone elsehaving the same trouble like me), that passing a topic name in local variable  to `lastWill` method is not going to do any good... 
Weird that I'm doing the same with `publish` and it seems fine.

Does not work
```
void test() {
	String lastwilltopic='some/topic';

	mqttClient.setKeepAlive(30).setWill(lastwilltopic.c_str(), 2, true, "0"); 
}
```

Does work
```
String lastwilltopic='some/topic';
void test() {
	mqttClient.setKeepAlive(30).setWill(lastwilltopic.c_str(), 2, true, "0"); 
}
```

By the way - perfect job. Thank you.

marvinroger, 2016-06-04:
I guess you do not have C/C++ background. :)
You don't call `connect()` in `test()`, but in another place, right?

When you call `setWill()`, the first parameter is a *pointer* to the string `some/topic`. The string is not passed *by value*, but instead by a pointer pointing to the address where `some/topic` is stored.
The problem is, when `test()` is done, the local variables are destroyed, so `lastwilltopic` won't exist anymore and `some/topic` will also be freed. Then, when you call `connect()`, AsyncMqttClient will attempt to access the string pointed by the pointer given previously to `setWill()`, but the string does not exist anymore, resulting in an unexpected behavior.

The second case does work, because `lastwilltopic` is in the global scope, so it won't be destroyed.

`publish()` works fine in both cases, because a publish is sent immediately, there is no need to store the pointer to the payload, so when the string is destroyed it does not matter, because it has already been sent.

drejkus, 2016-06-04:
Your guess is right, I'm not really good in C/C++. But i actually call `connect()` right after `setWill()` in the same context. The touble could have been on reconnecting... Never mind, lesson learned.

marvinroger, 2016-06-04:
So yes, the problem is probably on reconnections. Note, the issue you've encountered might also happen with `setClientId()`, `setCredentials()` and `setServer()`.

closed 2016-06-04 by drejkus

pedros89, 2022-04-21:
Hello, 
I had the same problem with local variables and lastWill and I made it work as you explained. You need a predefined global char array if the topic could change and you can populate it with LASTWILL_TOPIC.toCharArray(arrayWill, 64);
In the example the strings `rootTopic` and `childTopic` could be defined elsewhere globally in the code and be subjected to change. Now it works like a charm.

```
char arrayWill[64];

void launchMQTT() {
	
	mqttReconnectTimer = xTimerCreate("mqttTimer", pdMS_TO_TICKS(2000), pdFALSE, (void*)0, reinterpret_cast<TimerCallbackFunction_t>(connectToMqtt));
	
	const String LASTWILL_TOPIC = rootTopic + "/" + childTopic +  "/conn";
	LASTWILL_TOPIC.toCharArray(arrayWill, 64);
	mqttClient.setKeepAlive(30).setWill(arrayWill, 2, true, "<par name=\"conn\">0</par>");	//last will topic means that if ESP disconnects from broker, the broker will set this topic so
	Serial.print(arrayWill);

	mqttClient.onConnect(onMqttConnect);
	mqttClient.onDisconnect(onMqttDisconnect);
	mqttClient.onSubscribe(onMqttSubscribe);						
	mqttClient.onUnsubscribe(onMqttUnsubscribe);
	mqttClient.onPublish(onMqttPublish);							
	mqttClient.onMessage(onMqttMessage);
	mqttClient.setServer(MQTT_HOST, MQTT_PORT);
	mqttClient.setCredentials(MQTT_USERNAME, MQTT_PASSWORD);		
	
	connectToMqtt();
}
```