marvinroger/async-mqtt-client #6: using Async connection for a Server - Clinets connection

mkeyno, 2016-06-07:
hi  @marvinroger  sorry to make the issue for my question , actually I've planned to set a ESP module as AP and other ESP module as Station mode and like IOT home , the server received and send command to those , Iggre told me I should use asyinc lib instead of Esp8266WebServer lib , I really appreciated if you could guide me to what strategy should follow

marvinroger, 2016-06-08:
MQTT needs a broker to work, this project is a client. What you're asking is doable if someone implements a MQTT broker on the ESP8266, though.

mkeyno, 2016-06-08:
thanks for quick reply , Actually I intend to make sketch for both AP and Station mode , in AP mode , modules converted  to asysnc webserver which can gather data form all other station and probably send the command or acknowledgment to them , I think in simple home IOT, number of device or sensor wont exceed than 10 modules and async webserver can handle that, but I'm don't know how to install MQTT service for both mode , also in setup configuration I should make rule that each station introduce itself to AP module and register its IP and all sensor info that attached to it
during this post I reviewed the ESPeasy repo, they make very pretty IOT solution but yet can't figure out how to add other ESP module to AP module

marvinroger, 2016-06-08:
Your AP needs to be configured as a MQTT *broker*, this project is a *client* which needs a broker to communicate with (which is out the scope of this project).

mkeyno, 2016-06-08:
probably do you know such project or lib that go such way (MQTT broker)

marvinroger, 2016-06-08:
Sorry, I don't!

eos1d3, 2016-06-10:
You can use Mosquitto - An Open Source MQTT v3.1/v3.1.1 Broker
http://mosquitto.org/

It runs on many different platforms and include Raspberry PI.

i just tested it on Raspberry PI 3 and it works very well with this async client library.

marvinroger, 2016-06-20:
I thought you were asking for a MQTT broker on the ESP8266, but yes, mosquitto is the *leading* (I'm not sure if it is an "official" MQTT project).

Glad you achieved what you want.

closed 2016-06-20 by marvinroger