marvinroger/async-mqtt-client #7: Retained flag missing from received messages

eos1d3, 2016-06-11:
From http://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages

> Also the subscribing client can identify if a received message was a retained message or not, because the broker sends out retained messages with the retained flag still set to true. A client can then decide on how to process the message.

The receive message should include the retain flag. But the callback from current implementation do not have it. So there is no way to check if the message is a normal message or retained one.

marvinroger, 2016-06-11:
Right, I'll change the signature of the callback for a single AsyncMqttMessage parameter that contains every properties, otherwise the callbacks would be too heavy.

eos1d3, 2016-06-11:
Great! 

By the way, I am confused with the callback name of onPublish. I know why you use onPublish. But I think using `onSubscribeMessage` may be better.

marvinroger, 2016-06-11:
The callback names are based on the [libmosquitto ones](http://mosquitto.org/man/libmosquitto-3.html#idm116081070729792). `onPublish` means *when a publish we just sent is received by the broker*. I don't see how `onSubscribeMessage` illustrates this?

eos1d3, 2016-06-11:
Yes, I knew why you choose that name. I can fully understand. It is from a Sender to Broker point of view. It is definitely no problem.

But if the client is a just a Receiver, the condition is _broker notifies receiver that there is new message event_. So in receiver point of view, it is "subscribe  message" event.

It depends on the point of view. But keeping consistence with mosquitto is also good. This is just a small matter.

The callback will have all properties later is nice.

Thanks!!

marvinroger, 2016-08-28:
Check out the new FullyFeatured example.

closed 2016-08-28 by marvinroger

eos1d3, 2016-08-28:
I am not home and will test it a few days later.

commit 597ea146 (2016-08-28): Change onMessage signature to allow dup and retain flags - fixes #7