marvinroger/async-mqtt-client #8: Constant disconnect/reconnect cycle

pgebhard, 2016-06-20:
I was having some issues with exceptions occurring when I integrated this client into my own code, so I decided to try the example code provided in this repo. Unfortunately, using the provided example code, I am just seeing the following console output:

Connecting to Wi-Fi...... OK
Connecting to MQTT...
** Disconnected from the broker **
Reconnecting to MQTT...
** Disconnected from the broker **
Reconnecting to MQTT...
** Disconnected from the broker **
Reconnecting to MQTT...
** Disconnected from the broker **
Reconnecting to MQTT...

The connection never actually seems to be made to the broker even though I know it is running.

Any clues or tips?

jpmens, 2016-06-20:
That sounds a bit as though you're being disconnected due to authorization issues; check the broker logs to see whether that is the case.

marvinroger, 2016-06-20:
In the `onMqttDisconnect` event handler, add `Serial.println(static_cast<uint8_t>(reason));`

What's the number printed?

pgebhard, 2016-06-20:
For one, I've never seen any output in the broker log at /var/log/mosquitto/mosquitto.log even though the config file specifies that as the log_dest and log_type is set to all. Is there another config setting to be set?

As for the printed number, it's 0.

I had been using the PubSubClient in the past, and that connected fine using a single ESP8266. Once I had built a second device and had that start to connect to the broker, I started to get a similar type of disconnect/reconnect cycle. I thought it was an issue with the PubSubClient. For a single device connection using PubSubClient, though, it worked fine with my current Mosquitto configuration.

jpmens, 2016-06-20:
Re logging: off topic here, but that could be a permission problem on the file system.

Another thing: are you sure you don't have two clients with the same MQTT clientID connecting? They'd kick eachother out.

pgebhard, 2016-06-20:
For logging, yes, it appears that it was a permissions issue that was preventing writing to the log. I now see the initial startup lines from the broker, but I don't see anything related to the device trying to connect.

As for the potential clientID conflict, that could be the issue showing up when I was using PubSubClient. I will fix that and see if the two devices can connect simultaneously.

For my testing using AsyncMqttClient, however, I only have one device powered on for testing.

pgebhard, 2016-06-20:
Ok, count me as a n00b, but I found a typo in the broker server IP.

Thank you so much for your assistance, and I'm sure it will help me more quickly resolve any future issues I face.

Also, thank you for both of your great work on your many MQTT-related projects. I am following them and hope to contribute back when I can.

AndreasSteibl, 2017-05-03:
Hello i have the same problem.
On the ESP8266 i get following serial output (the 0 in the disconnect is the reason i got)
`** Connected to the broker **
Session present: 1
Subscribing at QoS 2, packetId: 1
Publishing at QoS 0
** Subscribe acknowledged **
  packetId: 1
  qos: 2
** Disconnected from the broker ** 
0
Reconnecting to MQTT...
`

On the raspberry (mosquitto Broker) i can see in log 
`New connection from 192..... on port 1883
Client lightWZ disconnected
client connected from .... as lightWZ (c0, k5)
Socket error on client lightWZ, disconnecting.`

What i am doing wrong?
I only have i device at the moment, with the pubSubClient all worked ok.