marvinroger/async-mqtt-client #12: Compilation error if this library is not used in .ino file

eos1d3, 2016-06-25:
I found that this library must be used and included only in main .ino file. 

I created a new sketch together with a cpp (mqtt.cpp) file. In cpp file I want to do all this mqtt tasks.
These are the only two lines of codes in cpp file:

```
#include <ESP8266WiFi.h>
#include <AsyncMqttClient.h>
```

And I always get warning at the second line:
```

Arduino: 1.6.9 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 921600, 4M (3M SPIFFS)"

In file included from C:\Users\eos1d3\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.h:25:0,

                 from C:\Users\eos1d3\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39,

                 from sketch\mqtt.cpp:2:

c:\users\eos1d3\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':

C:\Users\eos1d3\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token

 #define max(a,b) ((a)>(b)?(a):(b))

                  ^

c:\users\eos1d3\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_bvector.h: In member function 'std::vector<bool, _Alloc>::size_type std::vector<bool, _Alloc>::_M_check_len(std::vector<bool, _Alloc>::size_type, const char*) const':

C:\Users\eos1d3\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token

 #define max(a,b) ((a)>(b)?(a):(b))

                  ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

```

If I move the codes in cpp to .ino file, it compiles fine.
And I also downgraded ESP8266 core to 2.2, the problem is still the same.

eos1d3, 2016-06-25:
By adding  Arduino.h  helps a little, 

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <AsyncMqttClient.h>
```

but there are more other errors:

```
Arduino: 1.6.9 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 921600, 4M (3M SPIFFS)"

sketch\mqtt.cpp: In function 'void MQTT_connect()':

mqtt.cpp:104: error: 'class AsyncMqttClient' has no member named 'onSubscribeAck'

   mqttClient.onSubscribeAck(onMqttSubscribeAck);

              ^

mqtt.cpp:105: error: 'class AsyncMqttClient' has no member named 'onUnsubscribeAck'

   mqttClient.onUnsubscribeAck(onMqttUnsubscribeAck);

              ^

mqtt.cpp:106: error: no matching function for call to 'AsyncMqttClient::onPublish(void (&)(const char*, const char*, uint8_t, size_t, size_t, size_t))'

   mqttClient.onPublish(onMqttPublish);

                                     ^

sketch\mqtt.cpp:106:37: note: candidate is:

In file included from D:\eos1d3\Codes\Arduino\libraries\async-mqtt-client-master\src/AsyncMqttClient.h:4:0,

                 from sketch\mqtt.cpp:3:

D:\eos1d3\Codes\Arduino\libraries\async-mqtt-client-master\src/AsyncMqttClient.hpp:45:20: note: AsyncMqttClient& AsyncMqttClient::onPublish(AsyncMqttClientInternals::OnPublishUserCallback)

   AsyncMqttClient& onPublish(AsyncMqttClientInternals::OnPublishUserCallback callback);

                    ^

D:\eos1d3\Codes\Arduino\libraries\async-mqtt-client-master\src/AsyncMqttClient.hpp:45:20: note:   no known conversion for argument 1 from 'void(const char*, const char*, uint8_t, size_t, size_t, size_t) {aka void(const char*, const char*, unsigned char, unsigned int, unsigned int, unsigned int)}' to 'AsyncMqttClientInternals::OnPublishUserCallback {aka std::function<void(short unsigned int)>}'

mqtt.cpp:107: error: 'class AsyncMqttClient' has no member named 'onPublishAck'

   mqttClient.onPublishAck(onMqttPublishAck);

              ^

exit status 1
'class AsyncMqttClient' has no member named 'onSubscribeAck'

```

marvinroger, 2016-06-26:
Functions names were renamed to match libmosquitto events, see [the example](https://github.com/marvinroger/async-mqtt-client/blob/master/examples/FullyFeatured/FullyFeatured.ino).

Does it solve your issue?

eos1d3, 2016-06-26:
Didn't notice your change most of method names. These names make sense for me.  :)

There is no error now. I will test it later.

marvinroger, 2016-06-26:
Regarding the `Arduino.h`, it is automatically included in `.ino` sketches, not in `.cpp` files, so this is not an AsyncMqttClient bug. :)

Closing as this issue is solved, feel free to open another issue if you need to.

closed 2016-06-26 by marvinroger