marvinroger/async-mqtt-client #9: Provide information on memory management

sticilface, 2016-06-25:
Could you maybe add some documentation about the memory management. 

How big is the buffer?  
max send and receive size? 
Is the buffer fixed, or dynamic? (i've had a peek at the code, but it is not so obvious), is the message even buffered by the lib, or is it held in memory by the async TCP code, until the ACK is received.  if so are there imitations on msg size. 
Can you send messages bigger than this buffer (i'm assuming it is 1460bytes)

How does this work on the receive side?  Receiving large messages? 

looking forward to switching :)

marvinroger, 2016-06-25:
The documentation is definitely not complete.

There are no *internal* buffer, because I haven't had any issues using the TCP buffer directly.

The max receive size is about 1460 bytes per call to your `onMessage` callback. But the amount of data you can receive is unlimited, as if you receive, say, a 300kB payload (such as an OTA payload), then your `onMessage` callback will be called about 200 times, with the according `len`, `index` and `total` parameters. See [how OTA over MQTT can be done](https://github.com/marvinroger/homie-esp8266/blob/a123ce5fe8df05bd7d034eb11da205ba28046b9b/src/Homie/Boot/BootNormal.cpp#L164-L201), don't forget the `Update.runAsync(true)`.

You can send data as long as you stay below the available TCP window (which is 3-4kB on the ESP8266 as @me-no-dev told me, correct me if I am wrong). The data is indeed held in memory by the async TCP code until ACK is received. There are currently no way on this library to retrieve this available TCP window space, but it should not be too much of an issue as most of the MQTT packets you will send are, what, 20 bytes long? If you want to stream images it might indeed be problematic, otherwise it won't. So no, for now you cannot send messages bigger than this window.

As a proof of its "bulletproofness", just try to subscribe to all topics of a public broker, such as `test.mosquitto.org:1883`, and see how much it can handle without any weaknesses. Just a tip, set a high Serial baud rate, otherwise the ESP8266 might crash if you try to print too much data, or just throttle what you print.

sticilface, 2016-06-25:
this is great.  and as i had hoped.  good work, and thanks for the explanation. 

I can now work on removing all the code i wrote to send mqtt messages async in my project.  

PS.  your link to platformio does not work, and i can't find it in the registry.

sticilface, 2016-06-25:
regarding the window size, i guess that a method similar to the async web client can be used here... but correct the need to send large messages is not perhaps very common.

me-no-dev, 2016-06-25:
nothing is kept in memory actually. message is discarded as soon as it's sent by the stack. If you want to send large messages, you can do it the same way that the web server does it :)

marvinroger, 2016-06-25:
Thanks for the info @me-no-dev :)
And regarding the sending of large messages, if the need arises, I'll (try to) implement it.

marvinroger, 2016-06-25:
Regarding pio, some breaking changes are going to happen soon, so I won't publish it until the API is quite stable.

sticilface, 2016-06-25:
may i ask what is pio? 

@me-no-dev by not in memory it is kept until the ACK right? or what happens if no ACK is received?

me-no-dev, 2016-06-25:
actually lwip might be keeping it in memory, but no clue for how long really. I have not looked at how lwip handles tcp errors and resends the packet (if it even does so). If you do not get ack for certain time (timeout in the TCP lib), you will get onTimeout callback to let you know that packet was not acked

marvinroger, 2016-06-25:
pio = platformio, but nevermind, I've just published the lib. :)

dranger003, 2017-06-05:
@marvinroger Regarding your following comment:

> But the amount of data you can receive is unlimited, as if you receive, say, a 300kB payload (such as an OTA payload), then your onMessage callback will be called about 200 times, with the according len, index and total parameters.

I noticed after the first callback (i.e. index > 0) the topic is no longer set in `onMessage()` for subsequent callbacks. Is that intended?

marvinroger, 2017-06-05:
It's not. Do you have a minimal example sketch to reproduce? Also, can you please tell me what topic you're publishing to?

dranger003, 2017-06-05:
Simple sketch just having the `onMessage` callback set then send anything over 1500 bytes through and you'll see on the second callback the topic won't be set the same as the first callback. The topic I used is `ESP_A4D63B/I/Upload`.

dranger003, 2017-06-05:
And I'm using `libmosquitto` on the other end for publishing the data.
http://mosquitto.org/api/files/mosquitto-h.html#mosquitto_publish

dranger003, 2017-06-05:
Well, my apologies. After testing a bit more I found out I had a call to `strtok` on topic within the `onMessage` callback so I figure you must be using the same topic buffer on subsequent calls and `strtok` had modified it from the first call.

marvinroger, 2017-06-06:
@dranger003 oh yes, indeed, memory is not cheap on the ESP8266, so we reuse everything we can

closed 2017-06-14 by marvinroger

marvinroger, 2017-06-14:
Released.

commit dab67789 (2017-06-14): :memo: Provide information on memory management
Closes #9
commit bbc84655 (2017-06-14): :racehorse: Make client safer when TCP buffer is filled (#48)

* :racehorse: Implement initial code for safer ACKs

* :racehorse: Make ACKs safer
Part of #46
(PUBACK, PUBREC, PUBREL, PUBCOMP)

* :racehorse: Make CONNECT and DISCONNECT packets saf

* :memo: Provide information on memory management
Closes #9

* :shirt: Fix lint