marvinroger/async-mqtt-client #13: Connection and disconnection issues

eos1d3, 2016-06-27:
Hi it is me again. :)

With the FullyFeatured example, using current version of the library, I find it needs very long time to get disconnection status after first batch of callback calls. onMqttDisconnect callbackup works. But It tooke around 8 to 12 minutes to get disconnect status when tested with Adafruit IO.

Then I used the above example to test with Mosquitto on my RPI3. And I never see onMqttDisconnect is called. This is the log in 45 minutes:

```
Connecting to Wi-Fi.............. OK
Connecting to MQTT...
== Connected
Subscribing, packetId: 1
Publishing at QoS 0
Publishing at QoS 1, packetId: 2
Publishing at QoS 2, packetId: 3
== Subscribe acknowledged
  packetId: 1
  qos: 0
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 3
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 1
== Publish acknowledged
  packetId: 2
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 2
== Publish acknowledged
  packetId: 3
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 3

```

But I remember the old library worked very well with these brokers.

And the fatal problem is after disconnection, there is no way to reconnect to broker (Adadruit or Mosquitto), even passing the credentials to mqttclient is also the same.

All tested are done with FullyFeatured example and my own Sketh example.   And both Adafruit IO an Mosquitto on RPI3 are tested.

I also tested with ESP8266 core 2.2.0 and 2.3.0. Both versions are having the same problem.

marvinroger, 2016-06-27:
How do you disconnect? With the `disconnect()` method? Or do you manually stop the broker?

eos1d3, 2016-06-27:
Nothing to do for `disconnect()`.  See this: [https://learn.adafruit.com/mqtt-adafruit-io-and-you/intro-to-adafruit-mqtt#pinging-the-server](url)

Adafruit IO will  disconnect client in 5 mins if client does not access or ping it in this period. To keep it connected, Adafruit MQTT library has a method call `ping(()` to keep server connected.

In **previous** FullyFeatured example,  after the program is started, it calls `publish `to send data to broker. I will see serial output with callback messages. Then after keep-alive time, the broker will disconnect mqtt client. And the `onMqttDisconnect` is called. The function `mqttClient.connect()` inside the callback will reconnect the server.

Finally after re-connection, `onMqttConnect()` is called again. So I will see another groups of callback messages.

The above operations will keep looping forever. This is what the previous library did with with Adadruit IO.

In current version, after program is started. I also see the first group of callback messages, but after 8 to 12 minutes, the serial output will display `== Disconnected:` But the Adafruit IO will never be connected again with `mqttClient.connect();`. I had waited for 5 hours today to see this.

From my own sketch, it will publish data to Adafruit every 1 min. After 8-12 mins later, I also see disconnect message. But I found no data will be published after that. I still see the publish messages in serial output. 

Right now, I updated ESPAsyncTCP to the newest version. I did a test twice and wait for 20 mins. Now even after Adafruit disconnection, the library cannot display the disconnect messag any more.

ps: If you do not have Adafruit IO account, you can create one. Or I can email FullyFeatured sketch example to you with the account name and key. Pls let know me when to send to you.

eos1d3, 2016-06-27:
With this library and ESPAsyncTCP  are the latest. I just changed `loop() ` in FullyFeatured to:
`
```
void loop() {
  delay(60000);
  mqttClient.publish("eos1d3/feeds/t0", 0, true, "test 1");
  Serial.println("Publishing at QoS 0");
}
```

This is what I got from serial output:

```
Connecting to Wi-Fi... OK
Connecting to MQTT...
** Connected to the broker **
Subscribing at QoS 2, packetId: 1
Publishing at QoS 0
Publishing at QoS 1, packetId: 2
Publishing at QoS 2, packetId: 3
** Publish acknowledged **
  packetId: 2
** Subscribe acknowledged **
  packetId: 1
  qos: 2
** Publish acknowledged **
  packetId: 3
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
** Disconnected from the broker **
Reconnecting to MQTT...
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0
Publishing at QoS 0

```

From Adafruit web site, I found after "** Disconnected from the broker **" message, no message was updated even I keep seeing "Publishing at QoS 0" again and again.

marvinroger, 2016-08-27:
Is it still an issue since #18?

eos1d3, 2016-08-27:
This was solved previously. Thank!

marvinroger, 2016-08-27:
Great! :wink:

closed 2016-08-27 by marvinroger