marvinroger/async-mqtt-client #14: Keep alive

bcatalin, 2016-06-29:
I am publishing every 1 sec 3 messages to mosquitto, one of them is having around 500-600 bytes and other two around 200 bytes each. 

Keep Alive on ESP is set with mqttClient.setKeepAlive(30) to be 30 sec ( default 15). On my mosquitto I have 120 seconds timeout for the connection so will be like 3-4 consecutive packets missed before the cleanup.

So every 120 sec I've got a disconnect from mosquitto because of not receiving the keepalive. If I don't transmit any packet I don't have problems with keepalive. 

Do you have any suggestion ?

marvinroger, 2016-06-29:
There is a problem somewhere.

Here is how the Keep Alive works:

**Server-side**

* The client sends its keep alive in the connection packet
* The broker stores it, and disconnects the client if it does not receive any traffic during the Keep Alive time it received from the client + 50%

**Client-side**

* If the client does not send anything to the broker during the Keep Alive time it sent in its connection packet, it sends a PingReq to prove it is still alive

So if you send a message every second, a PingReq won't be send, because there is no need to. So in your case, it means the publish does not actually publish anything.

Can you provide a stripped-down code that shows this bug?

marvinroger, 2016-06-29:
And how did you define a timeout in mosquitto? There is no such thing, AFAIK. It's the whole point of the Keep Alive.

bcatalin, 2016-06-29:
It is a timeout in mosquitto for clients and one for bridging in the context.c file in function mqtt3_context_init. Right the +50% exist in the mosquitto code, and I don't see any PING requests with the exception of not sending any traffic to mosquitto.

I am sending 3 messages per second ~1k of data.

For publishing concern, I am receiving the data with a mobile app that is subscribing the same broker.

On mosquitto I have 5 times this message ( I am subscribing to 5 topics)
[send_server.c:59-_mosquitto_send_suback]=>
[send_server.c]Sending SUBACK to ESP_A08A78
    [read_handle.c - mqtt3_packet_handle] RX: SUBSCRIBE
[send_server.c:59-_mosquitto_send_suback]=>
[send_server.c]Sending SUBACK to ESP_A08A78
Wifi Task  Free Heap on  is: 16136 time:401 ADC:90
Wifi Task  Free Heap on  is: 16136 time:406 ADC:89
Wifi Task  Free Heap on  is: 16136 time:411 ADC:89
Wifi Task  Free Heap on  is: 16136 time:416 ADC:89
[read_handle.c] mqtt3_packet_handle RX: PINGREQ time=418 sec 
TX:Sending PINGRESP to root.1467204161994
Wifi Task  Free Heap on  is: 16136 time:421 ADC:90
Wifi Task  Free Heap on  is: 12912 time:426 ADC:89
Wifi Task  Free Heap on  is: 16136 time:431 ADC:89
Wifi Task  Free Heap on  is: 16136 time:436 ADC:89
Wifi Task  Free Heap on  is: 16136 time:441 ADC:91
Wifi Task  Free Heap on  is: 16136 time:446 ADC:89
Wifi Task  Free Heap on  is: 16136 time:451 ADC:88
Wifi Task  Free Heap on  is: 16136 time:456 ADC:86
Wifi Task  Free Heap on  is: 16136 time:461 ADC:85
Wifi Task  Free Heap on  is: 16136 time:466 ADC:92
Wifi Task  Free Heap on  is: 16136 time:471 ADC:93
Wifi Task  Free Heap  is: 16136 time:476 ADC:93
[read_handle.c] mqtt3_packet_handle RX: PINGREQ time=478 sec 
TX:Sending PINGRESP to root.1467204161994
Wifi Task  Free Heap on is: 16136 time:481 ADC:94
Wifi Task  Free Heap on is: 16136 time:486 ADC:94
Wifi Task  Free Heap on is: 16136 time:491 ADC:94
Wifi Task  Free Heap on is: 16136 time:496 ADC:93
Wifi Task  Free Heap  on is: 16136 time:501 ADC:93
[loop.c] Client ESP_A08A78 has exceeded timeout, disconnecting.

So I can see that after 100 sec is disconnected. ( with mosquitto timeout on 60 sec and async-mqtt keepalive to 30 sec)

Every sec I've put a check on heap, wifi and mqtt so I have

sendMQTTUpdate HEAP: 21816 msg_count:291 
sendMQTTUpdate=>sendMQTTUpdate WiFi:3 MQTT:1
sendMQTTUpdate HEAP: 21816 msg_count:292 
sendMQTTUpdate=>sendMQTTUpdate WiFi:3 MQTT:1
sendMQTTUpdate HEAP: 21816 msg_count:293 
sendMQTTUpdate=>sendMQTTUpdate WiFi:3 MQTT:0   //----connection lost.
sendMQTTUpdate=>sendMQTTUpdate WiFi:3 MQTT:0
sendMQTTUpdate=>sendMQTTUpdate WiFi:3 MQTT:0

marvinroger, 2016-06-29:
Alright, can you provide the sketch you're running? So I can reproduce.

bcatalin, 2016-06-29:
Sketch is big 5000 lines of code but I am setting the mqttClient not in setup but only once in loop.

**global**

AsyncMqttClient mqttClient;

**one time only in loop not in setup**

 mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  mqttClient.onMessage(onMqttMessage);
  mqttClient.setServer(sysCfg.mqtt_host, sysCfg.mqtt_port);
  mqttClient.setKeepAlive(30).setCredentials(sysCfg.mqtt_user, sysCfg.mqtt_pass).setClientId(dev_name);
  Serial.println("Connecting to MQTT...");
  mqttClient.connect();

  sendUpdate.every(1 * 1000 , sendMQTTUpdate); 

in 

void onMqttConnect() 
{
    Serial.println("** Connected to the broker **");
    mqttClient.subscribe(t1, 0);
    mqttClient.subscribe(t2, 0);
    //generic topics to subscribe on
    mqttClient.subscribe(t3,0);
    mqttClient.subscribe(t4,0);
    mqttClient.subscribe(t5,0);
    mqttClient.subscribe(t6,0);
}

subscribe to 6 topics and every second in **sendMQTTUpdate** I am sending 3 messages on 3 topics where the payload is JSON data 800 bytes 200 bytes 200 bytes. 

Nothing else special.

bcatalin, 2016-06-29:
Moved to PubSubClient and I don't see the issue. Every 120 sec I see PINGREQ/PINGRESP. So far 5000 messages and no disconnect.

eos1d3, 2016-07-06:
The keepalive does not work properly for me too.

I tested it with Fullyfeatured example for client with  `mqttClient.setKeepAlive(30)`, MQTT server is Mosquitto in RPI3, everything is default.

I only changed this in Fullyfeatured example:

```
void loop() {
  if (millis() >= t0 + 30000) {
    Serial.println("30 seconds passed");
    t0 = millis();
  }
}
```

Sometimes there is disconnection after 120 seconds, sometimes 240 seconds and sometimes there is never!

See the log here:

```
Connecting to Wi-Fi... OK
Connecting to MQTT...
== Connected
Subscribing, packetId: 1
Publishing at QoS 0
Publishing at QoS 1, packetId: 2
Publishing at QoS 2, packetId: 3
== Subscribe acknowledged
  packetId: 1
  qos: 0
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 3
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 1
== Publish acknowledged
  packetId: 2
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 2
== Publish acknowledged
  packetId: 3
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 3
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
== Disconnected: 
Reconnecting to MQTT...
== Connected
Subscribing, packetId: 1
Publishing at QoS 0
Publishing at QoS 1, packetId: 2
Publishing at QoS 2, packetId: 3
== Subscribe acknowledged
  packetId: 1
  qos: 0
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 3
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 1
== Publish acknowledged
  packetId: 2
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 2
== Publish acknowledged
  packetId: 3
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 3
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
== Disconnected: 
Reconnecting to MQTT...
== Connected
Subscribing, packetId: 1
Publishing at QoS 0
Publishing at QoS 1, packetId: 2
Publishing at QoS 2, packetId: 3
== Subscribe acknowledged
  packetId: 1
  qos: 0
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 3
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 1
== Publish acknowledged
  packetId: 2
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 2
== Publish acknowledged
  packetId: 3
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 3
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
== Disconnected: 
Reconnecting to MQTT...
== Connected
Subscribing, packetId: 1
Publishing at QoS 0
Publishing at QoS 1, packetId: 2
Publishing at QoS 2, packetId: 3
== Subscribe acknowledged
  packetId: 1
  qos: 0
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 3
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 1
== Publish acknowledged
  packetId: 2
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 2
== Publish acknowledged
  packetId: 3
== Publish received
  topic: eos1d3/feeds/lol
  qos: 0
  len: 6
  index: 0
  total: 6
payload: test 3
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
30 seconds passed
...

```

In my other sketch, packages are not published after keepalive time even it shows good result.

eos1d3, 2016-07-06:
And this was tested with Adafruit IO, with 60 seconds keepalive in client.
(sometimes there is disconnection, sometimes never)
```
Connecting to Wi-Fi..... OK
Connecting to MQTT...
== Connected
Subscribing, packetId: 1
Publishing at QoS 0
Publishing at QoS 1, packetId: 2
Publishing at QoS 2, packetId: 3
== Publish acknowledged
  packetId: 2
== Subscribe acknowledged
  packetId: 1
  qos: 0
== Publish acknowledged
  packetId: 3
One minute passed
One minute passed
One minute passed
One minute passed
One minute passed
One minute passed
One minute passed
One minute passed

```

marvinroger, 2016-08-28:
@eos1d3 once again, is it still an issue?

eos1d3, 2016-08-28:
Keepalive is not yet tested after reconnect bug. I will test and confirm
later.

flaviostutz, 2016-11-26:
See if https://github.com/marvinroger/async-mqtt-client/pull/31 will fix this. I made some enhancements in this are according to MQTT specs.

marvinroger, 2017-03-20:
Feel free to reopen if needed.

sticilface, 2017-05-21:
I'm looking into this as well.  I'm getting constant disconnected from my devices in Mosquitto logs

```
1495397753: New client connected from 192.168.1.152 as myDevice (c1, k5).
1495397754: Client myDevice disconnected.
1495397755: New connection from 192.168.1.152 on port 1883.
1495397755: New client connected from 192.168.1.152 as myDevice (c1, k5).
1495397763: Client myDevice disconnected.
1495397845: Client andrewblanket has exceeded timeout, disconnecting.
1495397845: Socket error on client andrewblanket, disconnecting.
1495397854: New connection from 192.168.1.187 on port 1883.
1495397854: New client connected from 192.168.1.187 as andrewblanket (c1, k5).
1495398521: Client andrewblanket has exceeded timeout, disconnecting.
1495398521: Socket error on client andrewblanket, disconnecting.
1495398526: New connection from 192.168.1.187 on port 1883.
1495398526: New client connected from 192.168.1.187 as andrewblanket (c1, k5).
1495398546: New connection from 192.168.1.207 on port 1883.
1495398546: Client myIR-livingroom already connected, closing old connection.
1495398546: Client myIR-livingroom disconnected.
1495398546: New client connected from 192.168.1.207 as myIR-livingroom (c1, k5).
1495398844: Client bedroom-ir has exceeded timeout, disconnecting.
1495398844: Socket error on client bedroom-ir, disconnecting.
1495398849: New connection from 192.168.1.104 on port 1883.
1495398849: New client connected from 192.168.1.104 as bedroom-ir (c1, k5).
1495399154: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
1495399209: Client tv has exceeded timeout, disconnecting.
1495399209: Socket error on client tv, disconnecting.
1495399214: New connection from 192.168.1.218 on port 1883.
1495399214: New client connected from 192.168.1.218 as tv (c1, k5).
1495399322: Client karoblanket has exceeded timeout, disconnecting.
1495399322: Socket error on client karoblanket, disconnecting.
1495399328: New connection from 192.168.1.186 on port 1883.
1495399328: New client connected from 192.168.1.186 as karoblanket (c1, k5).
1495400286: Client bedroom-ir has exceeded timeout, disconnecting.
1495400286: Socket error on client bedroom-ir, disconnecting.
1495400291: New connection from 192.168.1.104 on port 1883.
1495400291: New client connected from 192.168.1.104 as bedroom-ir (c1, k5).
1495400707: New connection from 192.168.1.207 on port 1883.
1495400707: Client myIR-livingroom already connected, closing old connection.
1495400707: Client myIR-livingroom disconnected.
1495400707: New client connected from 192.168.1.207 as myIR-livingroom (c1, k5).
1495400955: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
1495402399: Client karoblanket has exceeded timeout, disconnecting.
1495402399: Socket error on client karoblanket, disconnecting.
1495402404: New connection from 192.168.1.186 on port 1883.
1495402404: New client connected from 192.168.1.186 as karoblanket (c1, k5).
1495402756: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
1495402957: Client bedroom-ir has exceeded timeout, disconnecting.
1495402957: Socket error on client bedroom-ir, disconnecting.
1495402962: New connection from 192.168.1.104 on port 1883.
1495402962: New client connected from 192.168.1.104 as bedroom-ir (c1, k5).
1495404311: Client tv has exceeded timeout, disconnecting.
1495404311: Socket error on client tv, disconnecting.
1495404316: New connection from 192.168.1.218 on port 1883.
```

I'll try to debug it more, but its been going on for a while.  i've just not delved into the logs for a while.

bertmelis, 2017-05-28:
I would want to see this issue reopened as It experience the same as @sticilface.

Could find the bug however.

hakam80, 2017-05-31:
I am also facing same issue.....keep alive is missed from server side, and if I publish a command to the same ESP8266 that has no keepAlive, the ESP respond to that command. 
This means it is actually connected but the KeepAlive show that it is disconnected

marvinroger, 2017-05-31:
I still can't reproduce... I don't have a single timeout disconnection. Can you all use this exact version of ESPAsyncTCP and tell me if there's still an issue?

https://github.com/me-no-dev/ESPAsyncTCP/archive/3d030db489e263e16a5af43a095c83c12ed8e451.zip

Also, can you please try with the latest example?

marvinroger, 2017-06-01:
I finally was able to reproduce and I identified the cause: the TCP stack has some space available in its internal buffer. The client was not checking if there was enough space available to send packets; so it was trying to send content even though it was not able to, which leaded to the bug you're seeing.

Please all update to https://github.com/marvinroger/async-mqtt-client/commit/8bcd8fff4748546962fda9058d9a2915ba2a4c1e 

The Keep Alive will be sent whenever there's enough space available.

`publish`, `subscribe`and `unsubscribe` returns the packet ID. With the above change, it will return `0` if there's not enough space. (MQTT packet ID starts at 1 so there's no issue here). It is your responsability to retry the sending if you receive a `0`.

I'd be glad if you could try and report if the above commit fixed that or not!

marvinroger, 2017-06-01:
I had some reports from the homie-esp8266 repo: it works!

You can still report here and reopen if needed.

closed 2017-06-01 by marvinroger

sticilface, 2017-06-08:
Looks like it has worked.  I've had to update a lot of ESPs to be sure!  but the logs are not full of reconnects..  good job

NonaSuomy, 2017-06-28:
I think this is still going on for me.

Anything stands out in the code to you that maybe I'm not handling the connection properly?

https://gist.github.com/NonaSuomy/e50cfdf079fe1bbbcac30a381e34fd93

```
D: Metadata block 64 bytes
D: Streamtitle found, 55 bytes
D: StreamTitle='Ferry Corsten / Haris - Back To Paradise';
D: Publish to topic espradio/icy/streamtitle : Ferry Corsten / Haris - Back To Paradise
D: Publish to topic espradio/nowplaying : iHeart Trancid - Ferry Corsten / Haris - Back To Paradise
D: MQTT Publish acknowledged, packetId = 53
D: MQTT Publish acknowledged, packetId = 54
D: Metadata block 48 bytes
D: Streamtitle found, 38 bytes
D: StreamTitle='Hazem Beltagui - Nourey';
D: Publish to topic espradio/icy/streamtitle : Hazem Beltagui - Nourey
D: Publish to topic espradio/nowplaying : iHeart Trancid - Hazem Beltagui - Nourey
D: MQTT Publish acknowledged, packetId = 55
D: MQTT Publish acknowledged, packetId = 56
D: Publish to topic espradio/ip : 
D: MQTT Publish acknowledged, packetId = 57
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: No data input
D: No data input
D: Trying other station/file...
D: STOP requested
D: Song stopped correctly after 30 msec
D: New preset/file requested (7/0) from ihr/IHR_IEDM
D: Connect to new iHeartRadio host: IHR_IEDM
D: GET /api/livestream?version=1.9&mount=IHR_IEDMAAC&lang=en
...
D: Metadata block 48 bytes
D: Streamtitle found, 46 bytes
D: StreamTitle='Audien / Max - One More Weekend';
D: Recovered from dataloss
D: Metadata block 64 bytes
D: Streamtitle found, 51 bytes
D: StreamTitle='Morgan Page / Lissie - Don't Give Up';
D: Metadata block 48 bytes
D: Streamtitle found, 33 bytes
D: StreamTitle='Ofenbach - Be Mine';
D: Metadata block 64 bytes
D: Streamtitle found, 55 bytes
D: StreamTitle='Axwell & Ingrosso / Kid Ink - I Love You';
D: MQTT Connected to the broker 10.42.42.52, session is not present
D: Subscribing to espradio at QoS 0, packetId = 1 
D: Publish to topic espradio/icy/name : Evolution
D: MQTT Subscribe acknowledged, packetId = 1, QoS = 0
D: Publish to topic espradio/icy/streamtitle : Axwell & Ingrosso / Kid Ink - I Love You
D: Publish to topic espradio/icy/url : 
D: Publish to topic espradio/icy/genre : 
D: Publish to topic espradio/icy/sr : 
D: Publish to topic espradio/icy/content : audio/aacp
D: Publish to topic espradio/icy/bitrate : 48
D: Publish to topic espradio/nowplaying : Evolution - Axwell & Ingrosso / Kid Ink - I Love You
D: Publish to topic espradio/state : Playing
D: MQTT Publish acknowledged, packetId = 2
D: MQTT Publish acknowledged, packetId = 3
D: MQTT Publish acknowledged, packetId = 4
D: MQTT Publish acknowledged, packetId = 5
D: MQTT Publish acknowledged, packetId = 6
D: MQTT Publish acknowledged, packetId = 7
D: MQTT Publish acknowledged, packetId = 8
D: MQTT Publish acknowledged, packetId = 9
D: MQTT Publish acknowledged, packetId = 10
```

marvinroger, 2017-06-28:
Can you send the mosquito log?
Also, there might be another problem: your project is a webradio so it uses
a lot of bandwidth. Maybe the MQTT client has no chance to send the keep
alive packets because it does not have enough TCP buffer space available,
and there's nothing we can do to solve that, it's just that the ESP is not
powerful enough.

mbiato, 2017-06-28:
It was happening to me too. I had TLS enabled and I was setting the subscriptions in the onConnected callback. I had around 10 and the last ones returned 0 as packetId. While testing I removed the ones that failed. It keep disconnecting at the keepalive interval. The mosquitto log didn't show any PINGREQ packets. I could keep the connection alive if I manually sent messages to a topic from a computer. Could it be that it only process packets after ending the execution of the callbacks? I tried adding delays between subscriptions but they did nothing, I tried waiting with loops but the tcp buffer space available didn't change and eventually it resetted because of the watchdog. In the end, I just moved the subscriptions to the loop(), where delays worked, and left the onConnected callback just setting a boolean flag to trigger the subscriptions setup. I didn't understand why it didn't keep the connection alive even if the buffer available after setting the subscriptions in the onConnected callback was zero.

NonaSuomy, 2017-06-29:
Log:
```
1498704402: New connection from 10.13.37.168 on port 1883.
1498704402: New client connected from 10.13.37.168 as ESPRadio (c1, k15, u'OpenHAB').
1498704402: Sending CONNACK to ESPRadio (0, 0)
1498704403: Received SUBSCRIBE from ESPRadio
1498704403: 	espradio (QoS 0)
1498704403: ESPRadio 0 espradio
1498704403: Sending SUBACK to ESPRadio
1498704403: Received PUBLISH from ESPRadio (d0, q1, r0, m2, 'espradio/icy/name', ... (0 bytes))
1498704403: Sending PUBACK to ESPRadio (Mid: 2)
1498704403: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/name', ... (0 bytes))
1498704403: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icyname', ... (0 bytes))
1498704403: Received PUBLISH from ESPRadio (d0, q1, r0, m3, 'espradio/icy/streamtitle', ... (24 bytes))
1498704403: Sending PUBACK to ESPRadio (Mid: 3)
1498704403: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/streamtitle', ... (24 bytes))
1498704403: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icystreamtitle', ... (24 bytes))
1498704403: Received PUBLISH from ESPRadio (d0, q1, r0, m4, 'espradio/icy/url', ... (23 bytes))
1498704403: Sending PUBACK to ESPRadio (Mid: 4)
1498704403: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/url', ... (23 bytes))
1498704403: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icyurl', ... (23 bytes))
1498704403: Received PUBLISH from ESPRadio (d0, q1, r0, m5, 'espradio/icy/genre', ... (4 bytes))
1498704403: Sending PUBACK to ESPRadio (Mid: 5)
1498704403: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/genre', ... (4 bytes))
1498704403: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icygenre', ... (4 bytes))
1498704403: Received PUBLISH from ESPRadio (d0, q1, r0, m6, 'espradio/icy/sr', ... (0 bytes))
1498704403: Sending PUBACK to ESPRadio (Mid: 6)
1498704403: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/sr', ... (0 bytes))
1498704403: Received PUBLISH from ESPRadio (d0, q1, r0, m7, 'espradio/icy/content', ... (10 bytes))
1498704403: Sending PUBACK to ESPRadio (Mid: 7)
1498704403: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icysr', ... (0 bytes))
1498704403: Received PUBLISH from ESPRadio (d0, q1, r0, m8, 'espradio/icy/bitrate', ... (2 bytes))
1498704403: Sending PUBACK to ESPRadio (Mid: 8)
1498704403: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/bitrate', ... (2 bytes))
1498704403: Received PUBLISH from ESPRadio (d0, q1, r0, m9, 'espradio/nowplaying', ... (24 bytes))
1498704403: Sending PUBACK to ESPRadio (Mid: 9)
1498704403: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icybitrate', ... (2 bytes))
1498704403: Received PUBLISH from ESPRadio (d0, q1, r0, m10, 'espradio/state', ... (7 bytes))
1498704403: Sending PUBACK to ESPRadio (Mid: 10)
1498704413: Received PINGREQ from ESPRadio
1498704413: Sending PINGRESP to ESPRadio
1498704413: Received PINGREQ from ESPRadio
1498704413: Sending PINGRESP to ESPRadio
...
1498705002: Received PUBLISH from ESPRadio (d0, q1, r0, m15, 'espradio/ip', ... (0 bytes))
1498705002: Sending PUBACK to ESPRadio (Mid: 15)
...
1498705140: Received PINGREQ from ESPRadio
1498705140: Sending PINGRESP to ESPRadio
1498705142: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/CurrentDate', ... (19 bytes))
1498705145: Received PUBLISH from ESPRadio (d0, q1, r0, m16, 'espradio/icy/streamtitle', ... (31 bytes))
1498705145: Sending PUBACK to ESPRadio (Mid: 16)
1498705145: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/streamtitle', ... (31 bytes))
1498705145: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icystreamtitle', ... (31 bytes))
1498705145: Received PUBLISH from ESPRadio (d0, q1, r0, m17, 'espradio/nowplaying', ... (31 bytes))
...
1498705145: Sending PUBACK to ESPRadio (Mid: 17)
1498706423: Received PINGREQ from ESPRadio
1498706423: Sending PINGRESP to ESPRadio
1498704197: Client ESPRadio has exceeded timeout, disconnecting.
1498704197: Socket error on client ESPRadio, disconnecting.
```

bertmelis, 2017-06-29:
Keep in mind that they broker disconnects because of socket error. Not because of time-out (it get's logged differently, Assuming you're using mosquitto.)