marvinroger/async-mqtt-client #15: ssl support?

kenpeter, 2016-07-01:
Is it coming any time soon?

marvinroger, 2016-07-02:
I hope so! It has to be implemented in https://github.com/me-no-dev/ESPAsyncTCP, but I have no idea if/when it will. @me-no-dev maybe?

me-no-dev, 2016-07-02:
Hopefully soon :)

marvinroger, 2016-07-02:
@kenpeter it's the best ETA you will get. :)
Once it's implemented in the underlying TCP library, it will be a matter of days before it's usable in this library.

me-no-dev, 2016-07-03:
We are looking into ssl libs that can work asynchronously and that can be implemented in ESP8266 and ESP32 and have a few ideas, but nothing solid yet. Surely at some point in the near future the lib will support SSL, but I can not really say when exactly :)

marvinroger, 2016-07-03:
Thanks for the information :)

marvinroger, 2016-08-27:
SSL/TLS will be available for AsyncMqttClient when [esp8266/Arduino](https://github.com/esp8266/Arduino) 2.4.0 is released, thanks to @me-no-dev which implemented it in ESPAsyncTCP.

drejkus, 2016-11-08:
Can someone please point me in the right direction how to use SSL? Third party means using ESPAsyncTCP? An example would be perfect..

DavidM42, 2017-01-04:
Is there new information on how to use it with this library? I would also be interested to implement it into my project.

Nzbuu, 2017-01-15:
Looks like we're still waiting for esp8266/Arduino 2.4.0 to be released

s00500, 2017-01-23:
Puh this is a big drawback in the 2.x version... anyway this whole library is pretty awesome, it woud be great if the ssl support is working again

marvinroger, 2017-06-01:
The AsyncMqttClient will support SSL, it just cannot be implemented for the moment until 2.4.0 is released as it would be break the current code otherwise.

closed 2017-06-14 by marvinroger

marvinroger, 2017-06-14:
SSL has been implemented by @allanbartley. It will be released with v0.8.0 and will require esp8266/Arduino 2.4.0.

marvinroger, 2017-06-14:
Released!

kenpeter, 2017-06-14:
finally finally finally......

marvinroger, 2017-06-14:
I'd appreciate feedback, see if everything is working as it should 😊

allanbartley, 2017-06-14:
I'll re-test in a fresh project and do a quick example tonight.

allanbartley, 2017-06-15:
There are a couple of little bugs, which I'm working on a pull request for now.
There is also an issue where if the client is disconnected due to a bad fingerprint, there is a crash on attempting to reconnect. I haven't figured that one out yet. Should I raise an issue for this?

commit 0d20a539 (2017-06-14): :sparkles: Implement SSL support (#47)

* Added _secure private variable, initially set to false.
Added function to set value of secure.
Altered connect function to pass secure variable.
All modifications depend on ASYNC_TCP_SSL_ENABLED.

* Added ability to check server fingerprint against a known value during connection.
setServerFingerprint may be called multiple times to set more than one permitted value.

* Added check to make sure connection is set to secure before checking fingerprint on connect.

* Formatting changes.

* Added documentation.

* Fix lint.
Added note about platform for Pl [... cut]