marvinroger/async-mqtt-client #18: Library won't reconnect.

skorokithakis, 2016-07-28:
I have the library set to reconnect on disconnection, but that doesn't work, and I find the clients frequently disconnected even though they are on WiFi. Is there a setting I can use, or another way I can make them keep trying to reconnect?

eos1d3, 2016-07-28:
Confirmed that the reconnection never works for me. And keepalive also never works. These ruins the library to useless. Otherwise, this library would be nice. It has everything. But these two issues kill everything. 

I reported the issues for a while. Hope there is a chance to fix them.

skorokithakis, 2016-07-28:
Yeah, agreed. I'm thinking of either going back to the synchronous library, but I'll see if I can sidestep the issues by adding a call every 5 seconds to connect if not connected.

eos1d3, 2016-07-28:
I tested this with Adafruit IO and also Mosquitto. I remember even I publish every 60 seconds, there will finally a disconnection which breaks the connection permanently until restart. 
 Adafruit IO keepalive is 5 mins. So every 60 seconds is good to keep connection. But what I found is:
1. Sometimes there will be a disconnection with random time period.
2. Sometimes disconnection is never detected and the the program still works as if connection is still good. But nothing is actually published.

skorokithakis, 2016-07-28:
Ouch, the second kills it. I can deal with random disconnections if I have the script try to connect every 10-15 seconds, but if it doesn't even know it's disconnected, that's a no-go...

NeilSpp, 2016-07-29:
I have such a case, after the connection is disconnected  permanently until restart. I publish every 60 seconds to server.

marvinroger, 2016-07-29:
You're right, the library is not stable, that is why it is not actually released as v1. I've never had this kind of issue however. So you experience this behavior with the example sketch, for example?

We will obviously try to solve these issues, unfortunately I don't have ESPs at the moment for 2 weeks so I cannot do anything during this timelapse.

eos1d3, 2016-07-29:
Right! Just tested with Fullfeatured example with Adafruit IO and Mosquitto on PI3. Both have Keepalive and disconnection problem. Once disconnected, there is no way to reconnect until restart.

I am using NodeMCU which is actually ESP12E. And all related software/libraries are all updated to the newest version.

This library has good potential. Don't give up!

I can help if you find something to change. You can tell me what to try if you want.

marvinroger, 2016-07-29:
I won't give up, don't worry. :wink: 
Take a look at #1... This is weird! When I tested it, even with public brokers, it was very stable, even reconnections were stable. However I am not sure for the keep-alive.

Maybe an esp8266/Arduino or me-no-dev/ESPAsyncTCP update changed some things that we will need to figure out.

eos1d3, 2016-07-29:
For me, with the FullFeatured example, I will have two conditions:

1. The example program will detect disconnection at random time intervals, Sometimes 60 seconds, sometimes 5 mins.Once it detected, the library will try to reconnect, but it never gets connected.  I tried to publish at a fixed internal, when it gets disconnected, the publish never reports error. It looks like connection is good. But I see nothing from the broker.

2. The example program never detects the disconnection when connection is actually broken. I even tried to wait some hours and it just cannot resume.

skorokithakis, 2016-07-29:
You can see my usage here:

https://gitlab.com/stavros/sonoff/blob/08ac399becc01eb61666fdaa44c9daabd84d8e04/src/main.ino

marvinroger, 2016-08-04:
Can you test https://github.com/marvinroger/async-mqtt-client/pull/19?

kiralikbeyin, 2016-08-16:
Will be a fix very soon or i am planing to go with https://github.com/i-n-g-o/esp-mqtt-arduino

marvinroger, 2016-08-16:
The thing is I am not able to reproduce... I've been running a sample sketch for 72 hours, publishing at different rates, and no disconnection so far. So I am kind of stuck!

marvinroger, 2016-08-16:
When I restart my mosquitto instance, it reconnects just fine. I am pretty sure of the answer, but you're all using a modern mosquitto broker, right? I am using v1.4.9.

skorokithakis, 2016-08-16:
I'm using 1.4.8 :/ Too bad you can't reproduce this, it happened pretty regularly for me (disconnection after a few days, without reconnection).

eos1d3, 2016-08-16:
Right, I use everything (Arduino IDE, EPS8266 library, ESPAsyncTCP,
Mosquitto broker on PI3...) with the newest version.
And it NEVER can reconnect once.

kiralikbeyin, 2016-08-16:
I'm using mosca, i think this is not related with broker.

maybe we all use ESPAsyncWebServer except @marvinroger

eos1d3, 2016-08-16:
It may not be MQTT broker issue. The problem exists for both Adafruit IO and Mosquitto broker.

skorokithakis, 2016-08-16:
I switched the exact same code from async MQTT client to the normal, synchronous MQTT client and reconnections work fine.

marvinroger, 2016-08-16:
@kiralikbeyin said he was using ESPAsyncWebServer, are you @skorokithakis and @eos1d3? As it uses the same underlying ESPAsyncTCP, there might be a conflict or something.

skorokithakis, 2016-08-16:
Not that I know of. This is my code: https://gitlab.com/stavros/sonoff/blob/08ac399becc01eb61666fdaa44c9daabd84d8e04/src/main.ino

kiralikbeyin, 2016-08-16:
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>
#include <FS.h>
#include <Hash.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <StreamString.h>
#include <TimeLib.h>
#include <NtpClientLib.h>

#include "AsyncJson.h"
#include "ArduinoJson.h"
 #include <ESPAsyncWiFiManager.h>   

Should be a DNS problem

eos1d3, 2016-08-16:
I use this: https://github.com/me-no-dev/ESPAsyncTCP

I did try two versions of the above in July, both have the same problem.

One thing I cannot confirm, when I tested this library for the first time (at the time you released it), I remember I see a lot of disconnection and reconnection.  Then after changes of this and that, it never works now.

marvinroger, 2016-08-16:
Are you all using hostnames instead of direct IP?

skorokithakis, 2016-08-16:
I'm using a hostname, yes. You think resolution is failing?

eos1d3, 2016-08-16:
I use Mosquitto broker on PI3 at home with IP (192.168.2.xxx). It never works!

kiralikbeyin, 2016-08-16:
mqttClient.setServer(IPAddress(172, 20, 10, 7), 1883);
BUT
MDNS.begin(config.DeviceName.c_str());

eos1d3, 2016-08-16:
It can't be DNS problem as it can connect, public and subscribe normally for a while and then die. 
If it is DNS, it won't reach the broker.

eos1d3, 2016-08-16:
@marvinroger What is the date of your  ESPAsyncTCP library?

marvinroger, 2016-08-16:
@skorokithakis I don't know, I'm trying to isolate the issue
@eos1d3 the reconnection (not disconnection) issue might eventually come from DNS (it might work in the first place and not work in subsequent reconnects). I am just trying to gather the maximum amount of information to reproduce

The latest commit https://github.com/me-no-dev/ESPAsyncTCP/commit/6271c625061c8b55ce2b5c0a0f93f933d0fe8a20

On what OS is your broker running? Linux for @eos1d3, what about @kiralikbeyin and @skorokithakis ? Because I remember I had an issue with ports not being closed properly.

skorokithakis, 2016-08-16:
Ubuntu for me, although it's working fine with a different client library, so I'm not sure if it's the broker's issue, hmm.

eos1d3, 2016-08-16:
My Mosquitto is on PI3 Jessie (Debian Linux).
My test with Mosquitto used IP only.

kiralikbeyin, 2016-08-16:
@marvinroger iOS with mosca.

marvinroger, 2016-08-16:
@skorokithakis it might be since WiFiClient and ESPAsyncTCP are different on some points, notably the local port randomization. And I am on Windows! Can you try with this ESPAsyncTCP: https://github.com/me-no-dev/ESPAsyncTCP/archive/46cd942d3cc9fe84c2e606b4c4cf8ca9865356a3.zip

eos1d3, 2016-08-17:
Installed the above ESPAsyncTCP, using FullyFeatured example, with displaying "One minute passed" every one minute in loop(). Tested with Adafruit IO, the problem is there.

You can only see "Reconnecting to MQTT..." and no ""** Connected to the broker **".

```
Connecting to Wi-Fi........ OK
Connecting to MQTT...
** Connected to the broker **
Subscribing at QoS 2, packetId: 1
Publishing at QoS 0
Publishing at QoS 1, packetId: 2
Publishing at QoS 2, packetId: 3
** Publish acknowledged **
  packetId: 2
** Subscribe acknowledged **
  packetId: 1
  qos: 2
** Publish acknowledged **
  packetId: 3
One minute passed
** Disconnected from the broker **
Reconnecting to MQTT...
One minute passed
One minute passed
One minute passed
One minute passed
One minute passed
One minute passed
One minute passed
One minute passed
```

Will test with Mosquitto too.

eos1d3, 2016-08-17:
Mosquitto Broker on PI3
FullyFeatured example
above ESPAsyncTCP
loop changed to:

```
void loop() {
  if (millis() >= t0 + 30000) {
    Serial.println("30 seconds passed");
    t0 = millis();
    mqttClient.disconnect();
  }
}
```

Now, with more than 10 disconnections, it **CAN** always connect to broker!!

eos1d3, 2016-08-17:
Back to Adafruit IO example above, **keep everything unchanged**.
But changed 
`mqttClient.setServer("io.adafruit.com", 1883);`
to
`mqttClient.setServer(IPAddress(52, 70, 203, 194), 1883);`

Re-connection WORKS!

skorokithakis, 2016-08-17:
This may also have something to do with the fact that it tries to reconnect once, and, if that fails, it never tries again. It would be better if it could be set to reconnect on an interval.

eos1d3, 2016-08-17:
When to reconnect can actually be done by your application and just do nothing in `onMqttDisconnect`.

marvinroger, 2016-08-17:
Yes, I would like to keep the library as low-level as possible.

So the problem comes from https://github.com/me-no-dev/ESPAsyncTCP/commit/4edce1e9e538c040e029e50001cc4b496d509505 (before this commit: works on Linux/Unix only and after this commit: works on Windows only)

But @eos1d3 you're saying that DNS fails too, after subsequent reconnects?

marvinroger, 2016-08-17:
Anyway, @me-no-dev, we had a discussion about the local port randomization on Gitter (at which point you actually committed https://github.com/me-no-dev/ESPAsyncTCP/commit/4edce1e9e538c040e029e50001cc4b496d509505), but as said above it does not work fine in Linux/Unix env when disconnected not cleanly. WiFiClient works fine with both Windows and Linux servers, maybe you can borrow the randomization from there? (I am not sure I know what I am talking about though)

eos1d3, 2016-08-17:
Summaries for my tests:
1. When you use hostname, reconnection never works for 3 versions of ESPAsyncTCP with Adafruit IO.
2. For Adafruit IO and Mosquitto borkers, when using IP address, reconnection works for 3 versions of ESPAsyncTCP, including the latest one.
3. For all test, I use `mqttClient.disconnect()` to force a disconnection (without waiting for broker to disconnect)

So, my tests point to DNS problem. When using hostname, it can only connect once then no more for subsequent reconnects.

me-no-dev, 2016-08-17:
@marvinroger after the mentioned commit, the port randomisation works the same exact way as in the regular WiFiClient. Must be something else ;)

me-no-dev, 2016-08-17:
One question! Does the reconnect to dns hostname happen inside async callback?

marvinroger, 2016-08-17:
It does! I didn't think about this...

me-no-dev, 2016-08-17:
I know of this issue and it's on my to-do list to get it to work. Please file it on my repo so I can have it nagging me :) In a mean time, you can cache the IP of the broker (if given by dns) at init(your begin() or similar method called from setup() after WiFi is up) and use that IP in subsequent reconnects.

eos1d3, 2016-08-17:
I suggest waiting for real fix and using IP for the time being. Caching IP cannot help if ddns is used.

marvinroger, 2016-08-19:
Please try the latest ESPAsyncTCP!

eos1d3, 2016-08-19:
Using Adafruit IO with hostname, and use `mqttClient.disconnect()` to force disconnection.

Reconnect just works! :)

eos1d3, 2016-08-19:
Oh... got this:
```

** Disconnected from the broker **
Reconnecting to MQTT...

Exception (28):
epc1=0x402046cc epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd50 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffef0:  3ffea5d0 3ffea5b8 3ffea5bd 00000000  
3fffff00:  402260d0 00000000 3ffe8a70 00000000  
3fffff10:  3ffeea58 3fff0050 3ffefbe0 402238ec  
3fffff20:  3fff0054 3fff0050 04fa8acf 402239b2  
3fffff30:  3fff0054 3fff0050 04fa4e56 40105e3c  
3fffff40:  40104fee 04faa94f 60000600 40226279  
3fffff50:  40105e30 00000000 04faa94f 00000000  
3fffff60:  4021bd84 3ffee208 3ffee230 60000600  
3fffff70:  0feef2d4 3ffee230 3ffee208 4021bd91  
3fffff80:  4021bdd6 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee258 3fffdad0 3ffeeb38 402055b7  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<
bBÊsBrûÎ¦jòÿ

```

marvinroger, 2016-08-19:
Can you decode the stack dump? https://github.com/me-no-dev/EspExceptionDecoder/blob/master/README.md

@me-no-dev not totally fine yet!

eos1d3, 2016-08-19:
I want to test more. Now using disconnect with 15 seconds to see what will happen.

me-no-dev, 2016-08-19:
Let's see the stack decoded. Cant go more async than this :)

eos1d3, 2016-08-19:
I am currently getting problem with Adafruit IO. It is not your problem. I tried with PubSub MQTT client, it also has the same problem.

It is Adafruit IO sometimes won't let you connect.  Once you try to connect it, it will disconnect you immediately. Like this:

`
Connecting to Wi-Fi... OK
Connecting to MQTT...
** Disconnected from the broker **
Reconnecting to MQTT...
** Disconnected from the broker **
Reconnecting to MQTT...
** Disconnected from the broker **
Reconnecting to MQTT...
** Disconnected from the broker **
Reconnecting to MQTT...
** Disconnected from the broker **
Reconnecting to MQTT...
** Disconnected from the broker **
`

I found after you run a sketch with their Adafruit MQTT client library, Then the problem will be fixed for any non-Adafruit MQTT client libraries for a while. So it looks like their server is looking for some kind of signature.

Another woke around is to just ignore the errors and wait for a while, the problem will disappear. Any idea for this problem?

eos1d3, 2016-08-19:
I reported the problem in PubSubClient, one said: "Adafruit has added their own connack codes, they have their own broker." I don't know what it is.
See: https://github.com/knolleary/pubsubclient/issues/181

I think it is better to test with Moquitto with a DDNS name. I will try tomorrow.
Time to sleep now ;)

marvinroger, 2016-08-19:
I have no idea, the MQTT protocol is standardized, I don't see why Adafruit would change anything. Moreover it would not even work once if it was the case.

marvinroger, 2016-08-19:
https://io.adafruit.com/blog/

Code 6 : Client has exceeded connection rate limit. Please try again later.

So yes, you've disconnected too many times in a too short time window.

eos1d3, 2016-08-19:
I will change my router to forward the port to Mosquitto tomorrow so that I can use a hostname to test.

But currently I suspect it is the reentrance of onMqttConnect or onMqttDisconnect callbacks as I disconnect too fast in just 6 seconds.

marvinroger, 2016-08-19:
Yes this is the reason! Adafruit IO broker is throttled.

eos1d3, 2016-08-20:
Still using Adafruit IO to test, with 60 seconds disconnect interval. After running for while, I got this:

```
** Disconnected from the broker **
Reconnecting to MQTT...

Exception (28):
epc1=0x40204764 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd50 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffef0:  00000000 3fffdad0 3ffeeb38 00000030  
3fffff00:  402261d0 00000000 3ffe8a68 00000000  
3fffff10:  3ffeea58 3fff0050 3ffeff28 402239ec  
3fffff20:  3fff0054 3fff0050 16842732 40223ac2  
3fffff30:  00000000 1683f883 40207007 40105e3c  
3fffff40:  40104fee 168445b4 60000600 40226379  
3fffff50:  40105e30 00000000 168445b4 00000000  
3fffff60:  4021be84 3ffee208 3ffee230 60000600  
3fffff70:  480db416 3ffee230 3ffee208 4021be91  
3fffff80:  4021bed6 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee240 3fffdad0 3ffeeb38 402056b7  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  

```

And decode these got the followings:
```
Decoding 15 results
0x402261d0: tcpip_tcp_timer at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 81
0x402239ec: dns_check_entry at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/dns.c line 680
0x40223ac2: dns_tmr at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/dns.c line 313
0x40207007: pp_enable_noise_timer at ?? line ?
0x40105e3c: dns_timer at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 237
0x40104fee: wdt_feed at ?? line ?
0x40226379: sys_check_timeouts at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 420
0x40105e30: dns_timer at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 233
0x4021be84: ets_timer_handler_isr at ?? line ?
0x4021be91: ets_timer_handler_isr at ?? line ?
0x4021bed6: ets_timer_handler_isr at ?? line ?
0x402056b7: loop_task at C:\Users\andy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56

```

[36 more replies]

closed 2016-08-21 by marvinroger