marvinroger/async-mqtt-client #20: Feature request

kiralikbeyin, 2016-08-25:
Auto read a file and send response when got a "trigger message"  from broker?

marvinroger, 2016-08-27:
I don't understand what you mean, could you please clarify?

kiralikbeyin, 2016-08-27:
Broker to client -> "read a.json file and send me file content"
Esp-> open a.json file , read string, response to broker with file content.

Like mqttClient.publish("file/a.json", 0, true, "{"a":"json content"}" );

marvinroger, 2016-08-27:
AsyncMqttClient is intentionally low-level: it only provides raw MQTT stuff. What you're asking for can be achieved at the sketch level. This feature might be part of a higher level library, though, but I won't be the one creating this library, sorry.

closed 2016-08-27 by marvinroger