marvinroger/async-mqtt-client #22: Make a configurable limit on topic length

marvinroger, 2016-09-03:
The only way to crash the device currently is by sending a message on a topic with a length larger than what the ESP8266 RAM can handle. A `setMaxTopicLength()` function could be added, so the topic buffer would be allocated once, and the message would be skipped if the topic length is larger than the limit. What would be a good default? Can you guys tell me what's the longest topic you use?

eos1d3, 2016-09-06:
Hi,

I don't know why people will need very large length for this. I believe 32 to 128 bytes is more than enough. What do you think?

marvinroger, 2016-09-07:
For homie for example, topics can be like `base-topic-devices/firstfloor-bedroom-device/temperature/degrees/set`, which is 68 bytes already. So yes, I guess 128 bytes is a safe default.

marvinroger, 2016-10-10:
Implemented in 3f8cc2fe63dd9911e57a83ca0f2a8ffc8805bc16

closed 2016-10-10 by marvinroger