marvinroger/async-mqtt-client #23: Reconnect does not work when WIfi is lost

joao-ladeira, 2016-09-12:
If Wifi is lost, reconnect does not work, using the example provided (FullyFeatured)
The event "onMqttDisconnect" is fired but "onMqttConnect" is never trigered.

** Disconnected from the broker **
Reconnecting to MQTT...

Stays always like this.

marvinroger, 2016-09-28:
Can you update to the latest version of ESPAsyncTCP?

joao-ladeira, 2016-09-28:
Updated to the latest one but still the same.
Basically i'm just disconnecting my router and plugin it again as soon as i disconnect and connect the wifi router the example provided only outputs 

Reconnecting to MQTT...

once and then does nothing.
Using the pubsubclient from knolleary i have no problem, but since it has some limitations your lib seemed to be more appropriate. But until its stable enough i cant use it :/. 

When i have some time i will try to investigate this issue deeper.

marvinroger, 2016-09-28:
I'll test too! Thanks for the report.

flaviostutz, 2016-11-26:
Please see if https://github.com/marvinroger/async-mqtt-client/pull/31 will fix this. I made various tests like this one during its development.

osos, 2017-01-16:
I had the same problem.

Now running a test for several hours on master, with setalive, cleansession and setclient set to default, that is, setkeepalive=15 (not 5), setcleansession=true.

Seems to do the trick. Will report if otherwise detected

marxo789, 2017-01-27:
I also experience the same problem. Since I've downloaded all libs today, #31 doesn't seem to fix this issue. Also tried setkeepalive=15, setcleansession=true, without succes. 

When I shoot down broker or entire broker host, ESP8266 prints "Reconnecting to MQTT..." (as per FullyFeatured sample) repeatedly, when broker goes online, it reconnects fine. Only issue is, when I disable wifi on my router (simulating wifi loss). It prints "Reconnecting to MQTT..." only once, and never reconnects to wifi afterwards...

tinamore, 2017-01-29:
I had the same problem.

Reset router and lib can't connect to broken again.

marvinroger, 2017-03-20:
Are you sure you handle the wi-fi reconnect in your sketches?

tinamore, 2017-03-21:
@marvinroger Arduino esp8266 will Auto reconect wifi when lost, I think no need to handle the reconnect to wi-fi.

marxo789, 2017-03-21:
Okay, can you share code snippet, so I can try it? Since I never had issues with wifi reconnect with other libs, i presumed no extra handling is required. Thanks for help.

bertmelis, 2017-03-21:
Maybe this is related? https://github.com/esp8266/Arduino/issues/2186

It could be that the wifi is already connected (and by the async nature, this cuold be very well the case) when calling `WiFi.begin();` As I understand in the mentioned bu report, states are mangled up which could confuse Homie.

A workaround could be to change line 113 in BootNormal.cpp to
```c++
if (WiFi.status() != WL_CONNECTED) {
  WiFi.begin(Interface::get().getConfig().get().wifi.ssid, Interface::get().getConfig().get().wifi.password);
}
```
(only valid if the issue is related!)

marvinroger, 2017-03-21:
I'll try. 😉

marvinroger, 2017-03-21:
@bertmelis the issue doesn't happen on homie-esp8266 anyway, as Wi-Fi is handled differently

bertmelis, 2017-03-21:
Hell, I always confuse this repo with the Homie repo.
Anyway, after reading again, I don't think it's related as in the Arduino-issue, the ESP won't seem to connect at startup. Not sometime after several hours/days uptime.

nguyenhunga5, 2017-03-22:
I found problem in my esp, my solution is:

1. Check if wifi lost connect (alway after router reboot) -> call WiFi.disconnect(true); -> delay(500); ->then WiFi.begin("ssid", "pwd"); again.
2. After seconds (my config is 10s): check mqttClient->connected() == false -> call mqttClient->disconnect(); -> mqttClient->connect(); -> increase count and continues step 2. If count great than 10 times (my config is 10 times) -> reboot esp.

It worked for me. I hope help you.

thomasvnl, 2017-05-28:
Wanted to say that I am too experiencing this behaviour on all devices that use this library. Maybe the Full featured example should include how to deal with this, just like Homie does?

marvinroger, 2017-05-31:
Can you try this:

```c++
#include <ESP8266WiFi.h>
#include <Ticker.h>
#include <AsyncMqttClient.h>

#define WIFI_SSID "My_SSID"
#define WIFI_PASSWORD "My_password"

#define MQTT_HOST IPAddress(192, 168, 1, 10)
#define MQTT_PORT 1883

AsyncMqttClient mqttClient;
Ticker mqttReconnectTimer;

WiFiEventHandler wifiConnectHandler;
WiFiEventHandler wifiDisconnectHandler;
Ticker wifiReconnectTimer;

void connectToWifi() {
  wifiReconnectTimer.detach();
  Serial.println("Connecting to Wi-Fi...");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
}

void onWifiConnect(const WiFiEventStationModeGotIP& event) {
  Serial.println("Connected to Wi-Fi.");
  connectToMqtt();
}

void onWifiDisconnect(const WiFiEventStationModeDisconnected& event) {
  Serial.println("Disconnected from Wi-Fi.");
  mqttReconnectTimer.detach(); // ensure we don't reconnect to MQTT while reconnecting to Wi-Fi
  wifiReconnectTimer.attach(2, connectToWifi);
}

void connectToMqtt() {
  Serial.println("Connecting to MQTT...");
  mqttReconnectTimer.detach();
  mqttClient.connect();
}

void onMqttConnect(bool sessionPresent) {
  Serial.println("Connected to MQTT.");
  Serial.print("Session present: ");
  Serial.println(sessionPresent);
  uint16_t packetIdSub = mqttClient.subscribe("test/lol", 2);
  Serial.print("Subscribing at QoS 2, packetId: ");
  Serial.println(packetIdSub);
  mqttClient.publish("test/lol", 0, true, "test 1");
  Serial.println("Publishing at QoS 0");
  uint16_t packetIdPub1 = mqttClient.publish("test/lol", 1, true, "test 2");
  Serial.print("Publishing at QoS 1, packetId: ");
  Serial.println(packetIdPub1);
  uint16_t packetIdPub2 = mqttClient.publish("test/lol", 2, true, "test 3");
  Serial.print("Publishing at QoS 2, packetId: ");
  Serial.println(packetIdPub2);
}

void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
  Serial.println("Disconnected from MQTT.");

  if (WiFi.isConnected()) {
    Serial.println("Connecting to MQTT...");
    mqttReconnectTimer.attach(2, connectToMqtt);
  }
}

void onMqttSubscribe(uint16_t packetId, uint8_t qos) {
  Serial.println("Subscribe acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
  Serial.print("  qos: ");
  Serial.println(qos);
}

void onMqttUnsubscribe(uint16_t packetId) {
  Serial.println("Unsubscribe acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
}

void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {
  Serial.println("Publish received.");
  Serial.print("  topic: ");
  Serial.println(topic);
  Serial.print("  qos: ");
  Serial.println(properties.qos);
  Serial.print("  dup: ");
  Serial.println(properties.dup);
  Serial.print("  retain: ");
  Serial.println(properties.retain);
  Serial.print("  len: ");
  Serial.println(len);
  Serial.print("  index: ");
  Serial.println(index);
  Serial.print("  total: ");
  Serial.println(total);
}

void onMqttPublish(uint16_t packetId) {
  Serial.println("Publish acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  wifiConnectHandler = WiFi.onStationModeGotIP(onWifiConnect);
  wifiDisconnectHandler = WiFi.onStationModeDisconnected(onWifiDisconnect);

  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  mqttClient.onSubscribe(onMqttSubscribe);
  mqttClient.onUnsubscribe(onMqttUnsubscribe);
  mqttClient.onMessage(onMqttMessage);
  mqttClient.onPublish(onMqttPublish);
  mqttClient.setServer(MQTT_HOST, MQTT_PORT);

  connectToWifi();
}

void loop() {
}
```

I'll put this as the new example. This is the most bulletproof way of doing. 😉

thomasvnl, 2017-05-31:
I like it, will try this out tomorrow. Thanks Marvin!

marvinroger, 2017-05-31:
This is what I use and it works great!

thomasvnl, 2017-06-01:
@marvinroger Can confirm that it works like a charm indeed. Thanks again, Marvin 😉 !

marvinroger, 2017-06-01:
Awesome!

So yes, please don't use "blocking" Wi-Fi connection, it doesn't play well with async stuff.

closed 2017-06-01 by marvinroger

Gl0wf, 2022-05-29:
Hello. i have same problem but im using esp32 not like esp8266. Pls help me

Sergio-tix, 2023-02-21:
> 2\. After seconds (my config is 10s): check mqttClient->connected() == false -> call mqttClient->disconnect(); -> mqttClient->connect(); -> increase count and continues step 2. If count great than 10 times (my config is 10 times) -> reboot esp.

Can you please tell me how to implement this in code? How to check that at the current moment, right now, there is a connection with MQTT?
It doesn't work:
```
if (connected() == false)
       {
       }
```
How to write a function correctly?