marvinroger/async-mqtt-client #24: Non-clean sessions Support?

bickyb, 2016-09-28:
Hi,
I wanted to know if your MQTT supports non clean sessions.
In short, I am running my ESP8266 off of battery so I want it to periodically wake up and check for messages on the MQTT topic.   Since I am doing this, there is a chance I could miss messages that are published to the topic.  With support for non clean sessions, I could get the messages when my ESP8266 comes back online.

marvinroger, 2016-09-28:
It does not support it. The reason for this is the client needs to persist the following data:

> All messages in a QoS 1 or 2 flow, which are not confirmed by the broker
All received QoS 2 messages, which are not yet confirmed to the broker

We just cannot store the whole messages with such limited memory. We could enable non-clean sessions so that we can benefit from the advantages it brings, but then it would not be convention compliant as the client would not retain the messages not sent, as said above.

A simple workaround to that is, send your MQTT messages as retained, right?

bickyb, 2016-09-28:
Unfortunately with MQTT messages being retained, only the last, single message is stored by the broker.   So say for example, 5 messages are sent out to a topic while the client is off.  Only the last message would be received by the client when it comes back online.

I see the problem you are having with having to persist with a low amount of memory.  Would it be possible for you to enable non-clean sessions in the unconventional way you referred to?   It would still provide benefit to a number of users (like me).

marvinroger, 2016-09-28:
I think this is a good compromise. 😉

bickyb, 2016-09-28:
Yes, I think you would just need to document that supporting non clean sessions doesn't get you the full capability due to the memory restrictions of the Arduino.

Is this hard to implement in your library?

marvinroger, 2016-09-28:
It's actually easy, we only need to add a setter `setCleanSession()` and send the corresponding flag in the CONNECT packet.

bickyb, 2016-09-28:
Cool.  If you can let me know when you have added it, I can do some testing for you!

marvinroger, 2016-09-28:
Can you pull the latest git version and use `setCleanSession()`? I am totally unsure it will work, as I don't have an ESP8266 now.

marvinroger, 2016-10-07:
@bickyb did you have any chances to test ?

bickyb, 2016-10-11:
Hi,
From what I can see, it is not working.  I used mqttClient.setClientId("testclient"), and then used mqttClient.setCleansession(false).  I also subscribed a QoS 0.  
To test things out, I first subscribed to the broker, sent a message from my publisher, then unplugged my ESP8266 and sent a message while the ESP8266 was disconnected.  I then replugged the ESP8266 back in.  If it was working, I should have seen the message come across.  However, it did not.  Any thoughts?

marvinroger, 2016-10-11:
Can you try again with the latest git?

bickyb, 2016-10-11:
Hi,
I tried again with the latest and it did not seem to work.  I believe I am doing things correctly for my testing.

marvinroger, 2016-10-11:
Are stored in a session: all new QoS 1 or 2 messages, which the client missed while it was offline

Try sending and receiving at QoS 1 instead of 0.

bickyb, 2016-10-11:
Hi, to troubleshoot, I am publishing with QoS 1.   To do a sanity check, I used the mosquitto_sub client to see if the broker is retaining messages properly to handle offline clients and it seems to be working properly using the mosquitto_sub client.  Please check my email to you about trying to troubleshoot via a webex.

marvinroger, 2016-10-12:
Well, actually it works!

Update to the latest git (again) and flash the latest example sketch. On first boot, you will have a `Session present: 0`. If you reboot the ESP8266, you will see `Session present: 1`. This session present flag is sent by the broker, so it actually means the session got resumed.

Can you check?

bickyb, 2016-10-12:
Marvin,
I tried, but this time, the function call:
 mqttClient.onConnect(onMqttConnect);

seems to be failing compilation. 

sketch_oct11a:84: error: no matching function for call to 'AsyncMqttClient::onConnect(void (&)())'

   mqttClient.onConnect(onMqttConnect);

                                     ^
Here's my onMqttConnect method which was working before this last build:

void onMqttConnect() {
  Serial.println("** Connected to the broker **");
  uint16_t packetIdSub = mqttClient.subscribe("test", 1);
  Serial.print("Subscribing at QoS 1, packetId: ");
  Serial.println(packetIdSub);
}

marvinroger, 2016-10-13:
Yes, the signature of the onConnect callback now takes a sessionPresent
boolean, take a look at the latest example.

bickyb, 2016-10-13:
Ok,  I think we still have a problem.  
When my client connects, it appears to acknowledge that a a session is present:
Connecting to MQTT...
** Connected to the broker **
Session present: 1
Subscribing at QoS 1, packetId: 1
** Subscribe acknowledged **
  packetId: 1
  qos: 1
** Publish received **
  topic: test
  qos: 1
  dup: 0
  retain: 1
  len: 2
  index: 0
  total: 2

However, the message that I sent while the client was down was larger in length (15 characters).  The client does not report the 15 character message that I sent while the client was down.  However, it reports a 2 character message which I am not sure where it is getting it from.

marvinroger, 2016-10-15:
That's weird. Messages sent by the broker are sent in the exact same way with or without the clean session flag. I'll test next week, I don't have an ESP8266 at the minute.

marvinroger, 2016-10-16:
I tried to reproduce, and I think I know what happens on your side. The message you receive is not the one you sent while the client was down, it's a retained one (retain == 1, as shown on your logs). Try the following:

* `mosquitto_pub -t test -m "" -r -q 1`, this will clean the retained message on the topic
* Make sure your ESP8266 is disconnected
* `mosquitto_pub -t test -m "This is a test" -q 1`, this will send a non-retained message
* Power up your ESP8266
* You should receive a non-retained message of 14 bytes, which is what we expect (make sure the session present == 1)

So there's no issue on AsyncMqttClient, you just forgot you sent a retained message on the topic. 😉

marvinroger, 2016-10-20:
@bickyb any success?

bickyb, 2016-10-20:
Hi,
I just tried it out again and no success.  
One thing is to publish, I had to change things to:
mosquitto_pub -t test -m "This is a test" -q 1
to 
mosquitto_pub -t "test" -m "This is a test" -q 1
putting the topic name in quotes.

Note that 
mosquitto_pub -t "test" -m "" -r -q 1
did clean out the retained messages on the topic.

I know that Mosquitto is doing what it is supposed to as a I am using the Mosquitto command line client and get a message that was published prior to me starting the client:
I use the command "mosquitto_sub -c -i "dosclient" -q 1 -t "test"

Unfortunately, the arduino client does not pick up the messages that were missed when it was offline.  
I believe I am doing things right in my client.  I am attaching my sketch for you to take a look.

I think it might be more efficient to let you VNC into my machine to take a look and we can dedicate a time block for us to sync up.  Let me know if you would like to take that approach.

[sketch_oct11a.zip](https://github.com/marvinroger/async-mqtt-client/files/542195/sketch_oct11a.zip)

commit 035249cf (2016-09-28): Implement non-clean session support - fix #24
closed 2017-05-31 by marvinroger