marvinroger/async-mqtt-client #26: async-mqtt-client cannot compile with Arduino 1.16.11 and esp8266 2.3.0

kenpeter, 2016-10-12:
The error is:

> error: expected unqualified-id before '(' token
> #define max(a,b) ((a)>(b)?(a):(b))

I cannot remember the fix. It seems I need to redefine max in async-mqtt-client. Any advice?

[image]

[image]

marvinroger, 2016-10-12:
What's your sketch? See https://github.com/esp8266/Arduino/issues/398

karlp, 2016-11-22:
I had this, but fixed it by moving the include of ```<AsyncMqttClient.h>``` to "last"  (at least after some of the esp8266 headers)

I suspect asyncmqttclient.h should get some extra includes in it, so it doesn't depend on other libraries pulling in the right things.

marvinroger, 2016-12-03:
No, this is an esp8266/Arduino issue, see the issue linked above. I cannot do anything from async-mqtt-client, unfortunately.

marvinroger, 2017-03-20:
Closing it as we can't do anything on our side.

closed 2017-03-20 by marvinroger