marvinroger/async-mqtt-client #27: setServer to use hostname instead of IP address

hakam80, 2016-10-30:
AsyncMqttClient& setServer(IPAddress ip, uint16_t port)

Is it possible to use hostname instead of IP address?

marvinroger, 2016-10-30:
Of course, look at the API: https://github.com/marvinroger/async-mqtt-client/blob/master/docs/2.-API-reference.md

hakam80, 2016-10-30:
Sorry, I was blink from seeing the next command

hakam80, 2016-10-30:
Hi,
With IP address I can connect to mqtt broker, but with host name it is not connecting to mqtt broker, any advise?

#define MQTT_server IPAddress(100, 100, 1,110) //Ok, connection to broker established
#define MQTT_server "Bo19-srv" //Not working

hakam80, 2016-11-01:
Hi,
Any comment about why using hostname in MQTT_server is not working?

marvinroger, 2017-03-20:
Sorry for the late response.
A valid domain is an FQDN, `Bo19-srv` is not. AsyncMqttClient does not support mDNS and co, sorry!

closed 2017-03-20 by marvinroger