marvinroger/async-mqtt-client #29: panic in onMessage handler

karlp, 2016-11-22:
hi,  I'm getting a panic in my onMessage handler:

```
MQTT: got msg on topic: advent/aaa/c

Panic /home/karlp/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:98 __yield
< stack dump omitted, but I can provide it if you're keen >
```

The fail there is from "cont_can_yield" failing, which has a few issues reported, apparently from non-arduino context code calling yield?  I suppose this implies that the handlers are "non arduino" code?  So code like the following in a handler:

```
void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total)
{
    Serial.printf("MQTT: got msg on topic: %s\n", topic);
    for (int i = 0; i < 5; i++) {
        strip.SetPixelColor(1, RgbColor(40, 40, 0));
        strip.Show();
        delay(40);
        strip.SetPixelColor(0, RgbColor(0, 40, 40));
        strip.Show();
        delay(40);
    }
    Serial.println("here!")
```

Is simply unacceptable?  I had presumed that since "delay" == yield, I could do this quite happily, but I guess not?  I suppose I can setup handlers here instead...

marvinroger, 2016-11-22:
Indeed, you can't use yield or delay or any function that uses them inside the callbacks. That's because the code actually run *async* as the name implies, and these functions are synchronous.

karlp, 2016-12-01:
I've worked around this, but given this is a callback to _user_ code, I find it disconcerting at best that even delay(0) will cause a panic.  Surely it's the user's responsibility to make sure they don't take too long in their message handler.

My concern is that if I'm not allowed to do anything here, I need to _copy_ messages into a queue and reprocess them later, and that's kinda defeating the purpose of having a message handler.  At that point, the fact that this library is capable of handling more messages is somewhat moot.

If I _am_ meant to be processing my messages here, then it should be made to look like the real user code environment, ie, the panic() check shouldn't be triggered.

marvinroger, 2017-03-20:
This is how the async stack works: it does not process code in the "main thread". You can avoid `delay` anyway, see http://playground.arduino.cc/Code/AvoidDelay
Sorry, but we cannot do anything to solve that. 😕

closed 2017-03-20 by marvinroger