marvinroger/async-mqtt-client #30: _clear() in _onTimeout() callback

drejkus, 2016-11-25:
What should calling `_clear()` in `_onTimeout()` do or what does actually this timeout mean? Because I think it makes my connection not work after internet connectivity recovery. `_onPoll()` actually fires quite happily, but there is one `_onTimeout()` call in between which ruins it... It flags the connection as not connected (due to `_clear()`) and it never recovers again...  
Maybe it's not a bug or a bug somewhere else, but commenting out `_clear()` seems to work for me.
(But `_onTimeout()` is still called once in a while, probably after sending ping, weird)

drejkus, 2016-11-25:
Simulated network outage (packets to port 1883 dropped) . At about packet #1178 connection was allowed again. 
 
[image]

flaviostutz, 2016-11-26:
Maybe https://github.com/marvinroger/async-mqtt-client/pull/31 will fix this. Please test.

drejkus, 2016-11-27:
I'll test it soon and I guess it will work fine as it leaves the `_clear()`. But I suspect the real issue might be in ESPAsyncTCP as that is where the timeout calls come from...

AllanMar, 2016-11-28:
I've been having the same issues and looked into it a little bit.

It looks like what happens is after sometime MQTT can disconnect/timeout which causes problems due to the recent change in ESPAsyncTCP (b0e8d5630be1cdc0c3021947fcfcd0172e4ae6c9). Which tracks the unack len. This length is never cleared so once the first disconnect happens, all ACK's timeout from that point on (since unack_len never reaches zero). This adds an ACK timeout delay to every send (5seconds), which I believe can then lead to the constant timeouts/reconnects we're seeing.  #31 seems to help, but I suspect that's just because it's trying to send keepalives more frequently than the original (0.7*keepalive). 

I'm trying resetting these counters in the _close is ESPAsyncTCP but i'll have to wait some time and see if this works out/solves the problem.

flaviostutz, 2016-11-28:
Great diagnosis. I will look into it if I have time. The problem was avoided due to the fact that now I disconnect simply by verifying ping req/res between client and server (if there is a timeout on tcp, it will appear on the ping mechanism too, so those tcp timeouts are being ignored for now).

In your tests is it working ok now?

AllanMar, 2016-11-28:
It works better, but the root issue here still seems to be that ACK's are getting lost. I'm going to gear up a sniffer to see if I can figure out what's going on here. The main problem seem's to be that these ack timeouts introduce a 5s delay before anything else is sent which seems to compound problems. Also once the first ACK fails, they all seem to after that until disconnecting.

The issue seems to be on the ESPAsyncTCP side, rolling back before the commits I mentioned previously has been my biggest improvement  so far.

flaviostutz, 2016-11-28:
That's very possible...

AllanMar, 2016-12-01:
So in investigating this issue (enabling sniffer)  I ended up updating my router and disabling hardware acceleration on it (not sure which may be responsible). Ever since this i'm not getting the lost ACK's so i'm not having any of the follow on problems I seen previously. I still believe how this is handled in the latest ESPAsyncTCP is not ideal but I'm probably not going to spend much time on it now that my root issue is resolved.

marvinroger, 2016-12-03:
Can you open an issue on ESPAsyncTCP, so that we can close this one?

marvinroger, 2017-03-20:
Closing as this is not an AsyncMqttClient issue.

closed 2017-03-20 by marvinroger

sticilface, 2017-05-04:
Can I ask what the status is of this... I'm getting constant timeouts on my ESP MQTTs... 
```
33046: New client connected from 192.168.1.224 as bathroomlights_1640e0 (c1, k15).
1493933049: New connection from 192.168.1.197 on port 1883.
1493933049: New client connected from 192.168.1.197 as bedroomlights_1640e0 (c1, k15).
1493933053: New connection from 192.168.1.103 on port 1883.
1493933053: New client connected from 192.168.1.103 as livingroomlamp_1640e0 (c1, k15).
1493933068: Client bathroomlights_1640e0 has exceeded timeout, disconnecting.
1493933068: Socket error on client bathroomlights_1640e0, disconnecting.
1493933071: Client bedroomlights_1640e0 has exceeded timeout, disconnecting.
1493933071: Socket error on client bedroomlights_1640e0, disconnecting.
1493933075: Client livingroomlamp_1640e0 has exceeded timeout, disconnecting.
1493933075: Socket error on client livingroomlamp_1640e0, disconnecting.
1493933076: New connection from 192.168.1.224 on port 1883.
1493933076: New client connected from 192.168.1.224 as bathroomlights_1640e0 (c1, k15).
1493933079: New connection from 192.168.1.197 on port 1883.
1493933079: New client connected from 192.168.1.197 as bedroomlights_1640e0 (c1, k15).
1493933083: New connection from 192.168.1.103 on port 1883.
1493933083: New client connected from 192.168.1.103 as livingroomlamp_1640e0 (c1, k15).
1493933098: Client bathroomlights_1640e0 has exceeded timeout, disconnecting.
1493933098: Socket error on client bathroomlights_1640e0, disconnecting.
1493933101: Client bedroomlights_1640e0 has exceeded timeout, disconnecting.
1493933101: Socket error on client bedroomlights_1640e0, disconnecting.
1493933105: Client livingroomlamp_1640e0 has exceeded timeout, disconnecting.
1493933105: Socket error on client livingroomlamp_1640e0, disconnecting.
1493933106: New connection from 192.168.1.224 on port 1883.
1493933106: New client connected from 192.168.1.224 as bathroomlights_1640e0 (c1, k15).
1493933109: New connection from 192.168.1.197 on port 1883.
1493933109: New client connected from 192.168.1.197 as bedroomlights_1640e0 (c1, k15).
1493933113: New connection from 192.168.1.103 on port 1883.
1493933113: New client connected from 192.168.1.103 as livingroomlamp_1640e0 (c1, k15).
1493933128: Client bathroomlights_1640e0 has exceeded timeout, disconnecting.
1493933128: Socket error on client bathroomlights_1640e0, disconnecting.
1493933131: Client bedroomlights_1640e0 has exceeded timeout, disconnecting.
1493933131: Socket error on client bedroomlights_1640e0, disconnecting.
1493933135: Client livingroomlamp_1640e0 has exceeded timeout, disconnecting.
1493933135: Socket error on client livingroomlamp_1640e0, disconnecting.
1493933136: New connection from 192.168.1.224 on port 1883.
1493933136: New client connected from 192.168.1.224 as bathroomlights_1640e0 (c1, k15).
1493933139: New connection from 192.168.1.197 on port 1883.
1493933139: New client connected from 192.168.1.197 as bedroomlights_1640e0 (c1, k15).
1493933143: New connection from 192.168.1.103 on port 1883.
```

first noticed with a QOS of 2 set on a device for sending the command.   This is happening every 5 seconds or so!  

Not sure what is going on really!  

My config consists of a will...
```
  mqttClient.setKeepAlive(5).setWill(will.c_str(), 2, true, "offline").setClientId(device_name.c_str());
```

is this the same problem as above?

marvinroger, 2017-06-02:
@sticilface fixed with the v0.7.0.