marvinroger/async-mqtt-client #33: String instead of char*

ionciubotaru, 2016-12-02:
Is it possible to use String instead of char* on .publish, .subscribe, .setwill
Strings are easier to use than char*. 

For example:
String topic=id+"/order"
mqttClient.publish(topic, 0, true, "1");

marvinroger, 2016-12-03:
```c++
String topic=id+"/order"
mqttClient.publish(topic.c_str(), 0, true, "1");
```

It's up to you whether you want to use String or not (which is kind of a bad practice in embedded env). This library is low-level, so it won't directly support Strings.

Note you should prefer static allocation. Is this really more complicated:

```c++
char topic[32];
const char* id = "myId";
strcpy(topic, id);
strcat(topic, "/order");
```

I don't think so. 😉

ionciubotaru, 2016-12-04:
I upload the xml file to MQTT server in order to configure the web page automatically for a device.
The controlling web page don't know how many devices is installed and which GPIO use each. It use the xml files uploaded by each ESP to auto configure.

```
String readGpioFile() {
  File configFile = SPIFFS.open("/gpio.txt", "r");
  String gpio="";
  if (!configFile) {
    Serial.println("Failed to open gpio file");
  } else {
    size_t size = configFile.size();
    if (size > 512) {
      Serial.println("Config file too large");
    } else {
      gpio = configFile.readString();
    }
  }
  configFile.close();
  return gpio;
}

void setup () {
//  connecting stuff to wifi and mqtt
  char topic[32];
  const char* id = "myId";
  strcpy(topic, id);
  strcat(topic, "/configFile");
  mqttClient.publish(topic, 1, true, readGpioFile());
}
```

Please help me with any suggestion

marvinroger, 2017-03-20:
See http://stackoverflow.com/questions/7383606/converting-an-int-or-string-to-a-char-array-on-arduino

This library is low-level, so we won't provide any high level abstraction like `String`, this might be a good use case for a higher level lib, however. 😉

closed 2017-03-20 by marvinroger