marvinroger/async-mqtt-client #34: OTA over MQTT

ionciubotaru, 2016-12-02:
Please consider adding OTA over MQTT - it is very useful when you have a large number of ESP modules  
I found un example here https://github.com/Imroy/pubsubclient/blob/master/examples/ESP8266-OTA/ESP8266-OTA.ino it works fine. Your library is better and I prefer to use it.

Nzbuu, 2016-12-02:
Is this what you're after https://github.com/marvinroger/homie-esp8266?

ionciubotaru, 2016-12-02:
Homie is a great library, very advanced for a newbie.
A small standalone example will be good for everybody.

marvinroger, 2016-12-03:
Advances for newbie, [really?](https://homie-esp8266.readme.io/docs)

marvinroger, 2017-03-20:
As the previously closed issue, this library is meant for the raw MQTT client. If you want some higher features like OTA over MQTT, see [Homie for ESP8266](https://github.com/marvinroger/homie-esp8266) and its source code.

closed 2017-03-20 by marvinroger

dvv, 2018-07-06:
@ionciubotaru A workaround could be:

```c
static void onMqttMessage(char *topic, char *payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {

  // update requested?
  if (strcmp(topic, MQTT_CLIENT "/update") == 0) {
    int _lastError = 0;
    // start update
    if (index == 0) {
      Update.runAsync(true);
      if (!Update.begin(total, U_FLASH)) {
        _lastError = Update.getError();
        Update.printError(Serial);
        // ignore the rest
        _topic[0] = 0;
      }
    }
    // do update in chunks
    if (!_lastError) {
      if (Update.write((uint8_t *)payload, len) != len) {
        _lastError = Update.getError();
        Update.printError(Serial);
        // ignore the rest
        _topic[0] = 0;
      }
    }
    // finish update
    if (!_lastError) {
      if (index + len == total) {
        if (!Update.end()) {
          _lastError = Update.getError();
          Update.printError(Serial);
        } else {
          // restart?
          ESP.restart();
        }
      }
    }
  }

  // ...
}
```

ionciubotaru, 2018-07-06:
Only @arendst can decide to include this new feature in Tasmota's code. From my point of view using OTA over MQTT allow user to disable web server and web client and decrease the size of the final bin file.
More, user need only one server (mqtt) not two (mqtt + http)