marvinroger/async-mqtt-client #36: Issues with Authentication

PfisterDaniel, 2016-12-22:
473/5000
Hello,
I use the following code to set the relay state of my Sonoffs device.

Https://bitbucket.org/xoseperez/espurna/src/8a21133ecf4bd34764b844eb68dea42e9276ef3a/code/src/mqtt.ino?at=master&fileviewer=file-view-default

But when I use user name and password, I get no connection to the MQTT broker.

If I use an MQTT broker without authentication everything works fine.

Maybe someone could help me?

Thank you very much

xoseperez, 2016-12-23:
Hi
I'm the developer at ESPurna. Apparently the reason is that the connection with authentication is not stable. I ran you example code without problem but when I use the same code in my firmware it disconnects very often from the broker.
I'm still debugging. If you have any ideas about what should I check, please let me know.

marvinroger, 2017-03-20:
Sorry for the response time.

Authentication works fine, I just tried. Make sure the username and password char arrays are still available even outside the scope where the `setCredentials` method is called. Something like this:

```c++
const char* username;

void function() {
  String test = "foobar";
  username = strdup(test.c_str());
```

Here, `username` will never be destroyed and auth will work.

closed 2017-03-20 by marvinroger