marvinroger/async-mqtt-client #37: async-mqtt-client conflict with ESPhttpUpdate.update("http://server/firmware.bin") ?

gorec2005, 2017-01-17:
thank you for you async client - it's cool!
but when i use 
t_httpUpdate_return ret = ESPhttpUpdate.update("http://server/firmware.bin");
and do update - update's debug say:
"Update failed error (-1): HTTP error: connection refused" and dod not make http connection to server!
how i can debug this?
and how i can disconnect and disable async-mqtt-client ? i did not found this in api...

marvinroger, 2017-03-19:
I guess sync code does not play well with async code. But you can disconnect the AsyncMqttClient with https://github.com/marvinroger/async-mqtt-client/blob/master/docs/2.-API-reference.md#void-disconnect

closed 2017-05-31 by marvinroger

gjt211, 2017-08-30:
Hi @marvinroger,
I too am having this problem. Has anyone looked into a way to make this work?
It is not practical to disconnect MQTT just to check if an update is available, and I don't think it's a good idea to just assume that MQTT is working and use that to checking if there is a firmware update available.
My sensors currently run the check every hour and if an update is found, they do the httpUpdate. This has been working fine for over 6 months, but now that I have moved to the Async libs (top work by the way!), this is one of the few issues I need to solve.

Any help other that suggesting I disconnect the AsyncMqttClient would be greatly appreciated.

gorec2005, 2017-09-06:
i just made small trick -  update via flag in main loop:
>void loop() {
>...
>  if (update==1){ vLog("Start update"); ESPhttpUpdate.update(url); update=0; }
>...
> }
flag(update) changed in the mqtt-callback...
and all work just fine for me