marvinroger/async-mqtt-client #39: big messages & order

ramarro123, 2017-03-19:
Hello,

i have a scenario where i receive different topic message with a big json (> 1024)

i have created a global buffer like
char mqttBuffer[MQTT_TOPICS][MAX_TOPIC_LEN];

and i concat the chunk that i receive on the callback onMqttMessage
to the correct index.

it's that approach correct?
the data it's there :) i can parse, read it and everything (when total = num_char_readed)
but it's that efficient? 

it consume a bit of memory, should i change it in some other way?

are the message chunk received in different order for different topic? 
ie: it's possible to reiceive a chunk of topic1, then topic2, then the rest of topic1?
if it's not possible, then i can just use 1 single buff, but i need a confirmation on that

marvinroger, 2017-03-19:
This approach is correct, 1k is not *that* much on the ESP8266. If you would like to reduce memory usage, you can use a streaming based JSON parser.

The TCP messages are received in the same order they was sent with. As MQTT packets are sent one-by-one with a single header on multiple TCP packets, yes, you can assume the order will stay the same. (1 single buff needed)

closed 2017-05-31 by marvinroger

arihantdaga, 2017-08-04:
I am also stuck with this. I am having messages of almost 3-4kb size. And i could not figure out, how to concat these messages. Can u share some piece of code ? I'll be very grateful.

marvinroger, 2017-08-04:
```c++

// outside of the callback:

std::unique_ptr<char[]> mqttPayloadBuffer;

// in the callback

if (mqttPayloadBuffer == nullptr || index == 0) mqttPayloadBuffer = std::unique_ptr<char[]>(new char[total + 1]); // empty the buffer

memcpy(mqttPayloadBuffer.get() + index, payload, len); // copy the content into it

if (index + len != total) return;  // return if payload buffer is not complete

// here, our mqttPayloadBuffer is compplete
```

arihantdaga, 2017-08-07:
@marvinroger . It works great. Thank you very much. :)

brunoudi, 2019-02-22:
> ```c++
> // outside of the callback:
>
> std::unique_ptr<char[]> mqttPayloadBuffer;
>
> // in the callback
>
> if (mqttPayloadBuffer == nullptr || index == 0) mqttPayloadBuffer = std::unique_ptr<char[]>(new [...]

This work perfectly to receive big messages. 

Is there something similar to send big messages? 10kb payload?