marvinroger/async-mqtt-client #42: null terminated payloads

sticilface, 2017-05-03:
I was wondering if it would be possible to make the buffer for the payload null terminated.  As in always have it len + 1, and have that be '\0'.  

I'm asking as for some reason sometime when I use mqtt spy i get a non null terminated string and it overruns the buffer and I risk an exception.  My way round it was to first copy into a new buyer that was null terminated.  Now I just set `buffer[len] = '\0'`, however I'm not sure how long you've made the receiving buffer so this might also not be the best practice.  

In the end I want to make it safe you use String(buffer), and avoid another intermediate copy! 

A

jpmens, 2017-05-04:
MQTT payloads are binary safe; String(buffer) makes sense only if you *know* it's a NUL-terminated string.

sticilface, 2017-05-04:
I realise this.  I've observed behaviour when sending just some strings using MQTT-spy that a null terminated string "off" is not actually null terminated...  So i was thinking more of a safety thing to just make the buffer n+1 = '\0' to ensure it always it!

RudyFiero, 2017-05-07:
I don't know if this is related but I wonder if the problem I am having is related. I'm using the example code, FullyFeatured.ino. In the onMqttMessage() function I added a few Serial.print lines so the payload would be printed.  What i get printed is the message plus a couple of characters. This happens for all but the last received topic in the example code. 

`//-------------------------------------------
void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {
  Serial.println("** Publish received **");
  Serial.print("  topic: ");
  Serial.println(topic);
  Serial.print("  qos: ");
  Serial.println(properties.qos);
  Serial.print("  dup: ");
  Serial.println(properties.dup);
  Serial.print("  retain: ");
  Serial.println(properties.retain);
  Serial.print("  len: ");
  Serial.println(len);
  Serial.print("  index: ");
  Serial.println(index);
  Serial.print("  total: ");
  Serial.println(total);
  
  Serial.print("payload --->");               //  added printing of payload
  Serial.print(payload);
  Serial.println("<---");  
  Serial.println("==========================================="); 
}
`
And what I get printed is the following.

`..... WiFi connected
Connecting to MQTT...
** Connected to the broker **
Session present: 1
Subscribing at QoS 2, packetId: 1
Publishing at QoS 0
Publishing at QoS 1, packetId: 2
Publishing at QoS 2, packetId: 3
** Subscribe acknowledged **
  packetId: 1
  qos: 2
** Publish received **
  topic: test/lol
  qos: 2
  dup: 0
  retain: 1
  len: 6
  index: 0
  total: 6
payload --->test 40<---
===========================================
** Publish received **
  topic: test/lol
  qos: 0
  dup: 0
  retain: 0
  len: 6
  index: 0
  total: 6
payload --->test 1@<---
===========================================
** Publish acknowledged **
  packetId: 2
** Publish received **
  topic: test/lol
  qos: 1
  dup: 0
  retain: 0
  len: 6
  index: 0
  total: 6
payload --->test 2P<---
===========================================
** Publish acknowledged **
  packetId: 3
** Publish received **
  topic: test/lol
  qos: 2
  dup: 0
  retain: 0
  len: 6
  index: 0
  total: 6
payload --->test 3<---
===========================================
`

test 3 prints out as I expected. The ones above have two additional characters at the end of the line. When I publish to the topic with mqtt-spy I don't get the additional characters.

marvinroger, 2017-05-31:
AsyncMqttClient almost does not allocate memory. What you get in the callbacks is the raw TCP buffer. If we add a null terminator, we overwrite the first byte of the next MQTT message, basically possibly overwriting the Length MSB byte, and this would broke the client as we would not be able to know the length of the payload.

So yes, the only way is to create a copy as you did in the first place.

closed 2017-05-31 by marvinroger