marvinroger/async-mqtt-client #44: Reconnect only works on second/third try....

sticilface, 2017-05-21:
Reconnecting always takes two goes.. 

Using this sketch, which disconnects after 10 seconds, and immediately reconnects..  you can see it has to try twice!  

here is the serial output, millis() in brackets...  forced disconnect and reconnect... 
```
Connecting to Wi-Fi........ OK
Connecting to MQTT...
 [    5080] Connected to the broker
 [    5081] Subscribe done

 [   14052] Manual Disconnect Started....
 [   14056] Disconnected from the broker reason = TCP_DISCONNECTED
 [   14056] Reconnecting to MQTT..
 [   14064] Connected to the broker
 [   14064] Subscribe done
 [   14093] Disconnected from the broker reason = TCP_DISCONNECTED
 [   14093] Reconnecting to MQTT..
 [   15102] Connected to the broker
 [   15102] Subscribe done
```

here is mosquito output.  initial connections first then 2 lines, and the disconnect and reconnect attempts. 
```
1495404891: New connection from 172.17.0.1 on port 1883.
1495404891: Client myDevice already connected, closing old connection.
1495404891: Client myDevice disconnected.
1495404891: New client connected from 172.17.0.1 as myDevice (c0, k5, u'username').

1495404900: Client myDevice disconnected.
1495404900: New connection from 172.17.0.1 on port 1883.
1495404900: Client myDevice disconnected.
1495404900: New client connected from 172.17.0.1 as myDevice (c0, k5, u'username').
1495404900: Socket error on client myDevice, disconnecting.
1495404901: New connection from 172.17.0.1 on port 1883.
1495404901: Client myDevice disconnected.
1495404901: New client connected from 172.17.0.1 as myDevice (c0, k5, u'username').
```

the sketch
```cpp
#include <ESP8266WiFi.h>
#include <AsyncMqttClient.h>

const char * ssid = "ccc";
const char * pass = "ccc";

AsyncMqttClient mqttClient;

void onMqttConnect(bool sessionPresent) {
  Serial.printf(" [%8u] Connected to the broker\n", millis());  
//  mqttClient.subscribe("test/a", 2);
//  mqttClient.subscribe("test/b", 2);
//  mqttClient.subscribe("test/c", 2);
//  mqttClient.subscribe("test/d", 2);
//  mqttClient.subscribe("test/e", 2);
//  mqttClient.subscribe("test/f", 2);
//  mqttClient.subscribe("test/g", 2);
  Serial.printf(" [%8u] Subscribe done\n", millis());  

  
}

void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
String text; 
  switch( reason) {
  case AsyncMqttClientDisconnectReason::TCP_DISCONNECTED:
     text = "TCP_DISCONNECTED"; 
     break; 
  case AsyncMqttClientDisconnectReason::MQTT_UNACCEPTABLE_PROTOCOL_VERSION:
     text = "MQTT_UNACCEPTABLE_PROTOCOL_VERSION"; 
     break; 
  case AsyncMqttClientDisconnectReason::MQTT_IDENTIFIER_REJECTED:
     text = "MQTT_IDENTIFIER_REJECTED";  
     break;
  case AsyncMqttClientDisconnectReason::MQTT_SERVER_UNAVAILABLE: 
     text = "MQTT_SERVER_UNAVAILABLE"; 
     break;
  case AsyncMqttClientDisconnectReason::MQTT_MALFORMED_CREDENTIALS:
     text = "MQTT_MALFORMED_CREDENTIALS"; 
     break;
  case AsyncMqttClientDisconnectReason::MQTT_NOT_AUTHORIZED:
     text = "MQTT_NOT_AUTHORIZED"; 
     break;
  
  }
  Serial.printf(" [%8u] Disconnected from the broker reason = %s\n", millis(), text.c_str() );
  Serial.printf(" [%8u] Reconnecting to MQTT..\n", millis());
  mqttClient.connect();
}

void onMqttSubscribe(uint16_t packetId, uint8_t qos) {
  Serial.printf(" [%8u] Subscribe acknowledged id: %u, qos: %u\n", millis() ,packetId, qos);
}

void onMqttUnsubscribe(uint16_t packetId) {
  Serial.println("** Unsubscribe acknowledged **");
  Serial.print("  packetId: ");
  Serial.println(packetId);
}

void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {
  Serial.println("** Publish received **");
  Serial.print("  topic: ");
  Serial.println(topic);
  Serial.print("  qos: ");
  Serial.println(properties.qos);
  Serial.print("  dup: ");
  Serial.println(properties.dup);
  Serial.print("  retain: ");
  Serial.println(properties.retain);
  Serial.print("  len: ");
  Serial.println(len);
  Serial.print("  index: ");
  Serial.println(index);
  Serial.print("  total: ");
  Serial.println(total);
}

void onMqttPublish(uint16_t packetId) {
  Serial.println("** Publish acknowledged **");
  Serial.print("  packetId: ");
  Serial.println(packetId);
}

uint32_t timer = 0;  

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  Serial.print("Connecting to Wi-Fi");
  WiFi.begin(ssid, pass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println(" OK");

  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  mqttClient.onSubscribe(onMqttSubscribe);
//  mqttClient.onUnsubscribe(onMqttUnsubscribe);
//  mqttClient.onMessage(onMqttMessage);
//  mqttClient.onPublish(onMqttPublish);
  mqttClient.setServer(IPAddress(192, 168, 1, 234), 1883);
  mqttClient.setKeepAlive(5).setCleanSession(false).setWill("topic/online", 2, true, "no").setCredentials("username", "password").setClientId("myDevice");
  Serial.println("Connecting to MQTT...");
  mqttClient.connect();
  timer = millis(); 
}

bool havetried = false; 

void loop() {
  if (!havetried && mqttClient.connected() &&  millis() - timer > 10000) {
    Serial.printf("\n [%8u] Manual Disconnect Started....\n", millis() ); 
    mqttClient.disconnect(); 
    timer = millis(); 
    havetried = true; 
  }
}
```

sticilface, 2017-05-21:
may or may not help.  but if i publish a load of messages first, then subscribe...  i get this

```
Connecting to Wi-Fi.. OK
Connecting to MQTT...
 [    1083] Connected to the broker
 [    1084] Subscribe done
 [    1105] Subscribe acknowledged id: 6, qos: 2
 [    1105] Subscribe acknowledged id: 7, qos: 2
 [    1105] Subscribe acknowledged id: 8, qos: 2
 [    1107] Subscribe acknowledged id: 9, qos: 2
 [    1111] Subscribe acknowledged id: 10, qos: 2
 [    1115] Subscribe acknowledged id: 11, qos: 2
 [    1119] Subscribe acknowledged id: 12, qos: 2
 [    1129] Publish acknowledged id: 1, qos: 1739
 [    1133] Publish acknowledged id: 2, qos: 1713
 [    1134] Publish acknowledged id: 3, qos: 83
 [    1137] Publish acknowledged id: 4, qos: 1163
 [    1141] Publish acknowledged id: 5, qos: 1483

 [   11051] Manual Disconnect Started....
 [   11059] Disconnected from the broker reason = TCP_DISCONNECTED
 [   11060] Reconnecting to MQTT..
 [   11069] Connected to the broker
 [   11070] Subscribe done
 [   11077] Subscribe acknowledged id: 6, qos: 2
 [   11077] Subscribe acknowledged id: 7, qos: 2
 [   11077] Subscribe acknowledged id: 8, qos: 2
 [   11078] Subscribe acknowledged id: 9, qos: 2
 [   11083] Subscribe acknowledged id: 10, qos: 2
 [   11087] Subscribe acknowledged id: 11, qos: 2
 [   11091] Subscribe acknowledged id: 12, qos: 2
 [   11096] Disconnected from the broker reason = TCP_DISCONNECTED
 [   11101] Reconnecting to MQTT..
 [   11115] Connected to the broker
 [   11115] Subscribe done
 [   11122] Subscribe acknowledged id: 6, qos: 2
 [   11122] Subscribe acknowledged id: 7, qos: 2
 [   11122] Subscribe acknowledged id: 8, qos: 2
 [   11124] Subscribe acknowledged id: 9, qos: 2
 [   11128] Subscribe acknowledged id: 10, qos: 2
 [   11132] Subscribe acknowledged id: 11, qos: 2
 [   11137] Subscribe acknowledged id: 12, qos: 2
 [   11145] Publish acknowledged id: 1, qos: 1942
 [   11150] Publish acknowledged id: 2, qos: 406
 [   11150] Publish acknowledged id: 3, qos: 784
 [   11154] Publish acknowledged id: 4, qos: 331
 [   11158] Publish acknowledged id: 5, qos: 565
```

which is a bit weird..

sticilface, 2017-05-22:
here it is with verbose logging in mosquito.  There seems to be a docker error on sending a SUBACK to the ESP... 

```
1495441755: New connection from 172.17.0.1 on port 1883.
1495441755: Client myDevice already connected, closing old connection.
1495441755: Client myDevice disconnected.
1495441755: New client connected from 172.17.0.1 as myDevice (c0, k60, u'username').
1495441755: Sending CONNACK to myDevice (1, 0)
1495441755: Received PUBLISH from myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441755: Sending PUBLISH to myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441755: Received PUBLISH from myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441755: Sending PUBLISH to myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441755: Received PUBLISH from myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441755: Sending PUBLISH to myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441755: Received PUBLISH from myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441755: Sending PUBLISH to myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441755: Received PUBLISH from myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441755: Sending PUBLISH to myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441755: Received SUBSCRIBE from myDevice
1495441755: 	test/a (QoS 0)
1495441755: myDevice 0 test/a
1495441755: Sending SUBACK to myDevice
1495441755: Received SUBSCRIBE from myDevice
1495441755: 	test/b (QoS 0)
1495441755: myDevice 0 test/b
1495441755: Sending SUBACK to myDevice
1495441755: Received SUBSCRIBE from myDevice
1495441755: 	test/c (QoS 0)
1495441755: myDevice 0 test/c
1495441755: Sending SUBACK to myDevice
1495441755: Received SUBSCRIBE from myDevice
1495441755: 	test/d (QoS 0)
1495441755: myDevice 0 test/d
1495441755: Sending SUBACK to myDevice
1495441755: Received SUBSCRIBE from myDevice
1495441755: 	test/e (QoS 0)
1495441755: myDevice 0 test/e
1495441755: Sending SUBACK to myDevice
1495441755: Received SUBSCRIBE from myDevice
1495441755: 	test/f (QoS 0)
1495441755: myDevice 0 test/f
1495441755: Sending SUBACK to myDevice
1495441755: Received SUBSCRIBE from myDevice
1495441755: 	test/g (QoS 0)
1495441755: myDevice 0 test/g
1495441755: Sending SUBACK to myDevice

1495441765: Received DISCONNECT from myDevice
1495441765: Client myDevice disconnected.
1495441765: New connection from 172.17.0.1 on port 1883.
1495441765: Client myDevice disconnected.
1495441765: New client connected from 172.17.0.1 as myDevice (c0, k60, u'username').
1495441765: Sending CONNACK to myDevice (1, 0)
1495441765: Received PUBLISH from myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441765: Sending PUBLISH to myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441765: Received PUBLISH from myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441765: Sending PUBLISH to myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441765: Received PUBLISH from myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441765: Sending PUBLISH to myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441765: Received PUBLISH from myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441765: Sending PUBLISH to myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441765: Received PUBLISH from myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441765: Sending PUBLISH to myDevice (d0, q0, r0, m0, 'test/a', ... (5 bytes))
1495441765: Received SUBSCRIBE from myDevice
1495441765: 	test/a (QoS 0)
1495441765: myDevice 0 test/a
1495441765: Sending SUBACK to myDevice
1495441765: Received SUBSCRIBE from myDevice
1495441765: 	test/b (QoS 0)
1495441765: myDevice 0 test/b
[... 40 lines cut ...]
1495441765: Received SUBSCRIBE from myDevice
1495441765: 	test/b (QoS 0)
1495441765: myDevice 0 test/b
1495441765: Sending SUBACK to myDevice
1495441765: Received SUBSCRIBE from myDevice
1495441765: 	test/c (QoS 0)
1495441765: myDevice 0 test/c
1495441765: Sending SUBACK to myDevice
1495441765: Received SUBSCRIBE from myDevice
1495441765: 	test/d (QoS 0)
1495441765: myDevice 0 test/d
1495441765: Sending SUBACK to myDevice
1495441765: Received SUBSCRIBE from myDevice
1495441765: 	test/e (QoS 0)
1495441765: myDevice 0 test/e
1495441765: Sending SUBACK to myDevice
1495441765: Received SUBSCRIBE from myDevice
1495441765: 	test/f (QoS 0)
1495441765: myDevice 0 test/f
1495441765: Sending SUBACK to myDevice
1495441765: Received SUBSCRIBE from myDevice
1495441765: 	test/g (QoS 0)
1495441765: myDevice 0 test/g
1495441765: Sending SUBACK to myDevice
```

marvinroger, 2017-05-31:
@sticilface thanks for the report! Can you test with the latest git rev?

closed 2017-05-31 by marvinroger

sticilface, 2017-06-05:
confirmed.  works correctly now. thanks for that

commit ce27b780 (2017-05-31): :bug: Fix doubled reconnections - closes #44