marvinroger/async-mqtt-client #45: Random disconnects cant figure out whats wrong.

NonaSuomy, 2017-05-27:
https://github.com/Edzelf/Esp-radio/issues/87

https://gist.github.com/NonaSuomy/e50cfdf079fe1bbbcac30a381e34fd93

```
D: Metadata block 48 bytes
D: Streamtitle found, 47 bytes
D: StreamTitle='Billy Talent - Afraid of Heights';
D: Publish to topic espradio/icy/streamtitle : Billy Talent - Afraid of Heights
D: MQTT Publish acknowledged, packetId = 34
D: 10 minutes over
D: Metadata block 64 bytes
D: Streamtitle found, 48 bytes
D: StreamTitle='The Killers - When You Were Young';
D: Publish to topic espradio/icy/streamtitle : The Killers - When You Were Young
D: MQTT Publish acknowledged, packetId = 35
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT reconnection attempt: 1
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
```

Any help would be appreciated.

For now, I just check if every loop if connected then reconnect but would like to find out what's causing the disconnect. MQTT broker is local and nothing else seems to have the disconnect happen I believe reason 0 is something to do with the client failing.

marvinroger, 2017-05-31:
Reason 0 means the TCP socket was closed. Do you have the MQTT broker log?

NonaSuomy, 2017-06-01:
```
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m45, 'espradio/icy/name', ... (0 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 45)
1496293893: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/name', ... (0 bytes))
1496293893: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icyname', ... (0 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m46, 'espradio/icy/streamtitle', ... (0 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 46)
1496293893: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/streamtitle', ... (0 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m47, 'espradio/icy/url', ... (0 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 47)
1496293893: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/url', ... (0 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m48, 'espradio/icy/genre', ... (0 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 48)
1496293893: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/genre', ... (0 bytes))
1496293893: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icystreamtitle', ... (0 bytes))
1496293893: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icyurl', ... (0 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m49, 'espradio/icy/sr', ... (0 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 49)
1496293893: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/sr', ... (0 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m50, 'espradio/icy/content', ... (0 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 50)
1496293893: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icygenre', ... (0 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m51, 'espradio/icy/bitrate', ... (0 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 51)
1496293893: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/bitrate', ... (0 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m52, 'espradio/nowplaying', ... (15 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 52)
1496293893: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icysr', ... (0 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m53, 'espradio/state', ... (7 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 53)
1496293893: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icybitrate', ... (0 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m54, 'espradio/icy/name', ... (0 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 54)
1496293893: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/name', ... (0 bytes))
1496293893: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icyname', ... (0 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m55, 'espradio/icy/url', ... (23 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 55)
1496293893: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/url', ... (23 bytes))
1496293893: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icyurl', ... (23 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m56, 'espradio/icy/genre', ... (4 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 56)
1496293893: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/genre', ... (4 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m57, 'espradio/icy/content', ... (10 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 57)
1496293893: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icygenre', ... (4 bytes))
1496293893: Received PUBLISH from ESPRadio (d0, q1, r0, m58, 'espradio/icy/bitrate', ... (2 bytes))
1496293893: Sending PUBACK to ESPRadio (Mid: 58)
1496293893: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/bitrate', ... (2 bytes))
1496293893: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icybitrate', ... (2 bytes))
1496293896: Received PUBLISH from ESPRadio (d0, q1, r0, m59, 'espradio/icy/streamtitle', ... (22 bytes))
1496293896: Sending PUBACK to ESPRadio (Mid: 59)
1496293896: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/streamtitle', ... (22 bytes))
1496293896: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icystreamtitle', ... (22 bytes))
1496293896: Received PUBLISH from ESPRadio (d0, q1, r0, m60, 'espradio/nowplaying', ... (22 bytes))
1496293896: Sending PUBACK to ESPRadio (Mid: 60)
1496293905: Received PUBLISH from ESPRadio (d0, q1, r0, m61, 'espradio/icy/streamtitle', ... (24 bytes))
1496293905: Sending PUBACK to ESPRadio (Mid: 61)
1496293905: Sending PUBLISH to OpenHAB2 (d0, q0, r0, m0, 'espradio/icy/streamtitle', ... (24 bytes))
1496293905: Received PUBLISH from OpenHAB2 (d0, q0, r0, m0, 'hq/state/icystreamtitle', ... (24 bytes))
1496293905: Received PUBLISH from ESPRadio (d0, q1, r0, m62, 'espradio/nowplaying', ... (24 bytes))
1496293905: Sending PUBACK to ESPRadio (Mid: 62)
1496293916: Received PINGREQ from ESPRadio
1496293916: Sending PINGRESP to ESPRadio
1496293927: Received PINGREQ from ESPRadio
1496293927: Sending PINGRESP to ESPRadio
1496293938: Received PINGREQ from ESPRadio
1496293938: Sending PINGRESP to ESPRadio
1496293948: Received PINGREQ from ESPRadio
1496293948: Sending PINGRESP to ESPRadio
1496293958: Received PUBLISH from ESPRadio (d0, q1, r0, m63, 'espradio/ip', ... (11 bytes))
1496293958: Sending PUBACK to ESPRadio (Mid: 63)
[... cut ...]
1496293968: Sending PINGRESP to ESPRadio
1496293979: Received PINGREQ from ESPRadio
1496293979: Sending PINGRESP to ESPRadio
1496293989: Received PINGREQ from ESPRadio
1496293989: Sending PINGRESP to ESPRadio
1496294000: Received PINGREQ from ESPRadio
1496294000: Sending PINGRESP to ESPRadio
1496294011: Received PINGREQ from ESPRadio
1496294011: Sending PINGRESP to ESPRadio
1496294033: Client ESPRadio has exceeded timeout, disconnecting.
1496294033: Socket error on client ESPRadio, disconnecting.
```

marvinroger, 2017-06-01:
This is a keep alive issue; duplicate of https://github.com/marvinroger/async-mqtt-client/issues/14, please watch this issue

closed 2017-06-01 by marvinroger