marvinroger/async-mqtt-client #51: esp8266 crashed....

gorec2005, 2017-06-23:
Hi!
Esp crashed when got retained message in subscribed topic - how i can debug it?

`Exception (28):
epc1=0x402073d3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys
sp: 3ffffc80 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe20:  3ffebe3e 0000002d 3fff1620 40207165
3ffffe30:  3fff3714 00000000 402066a8 4020b5e8
3ffffe40:  3fff411c 3ffffed0 40206660 4020b64
...
fffff80:  4020de06 3ffeef78 00000000 3fff3b6c
3fffff90:  3fffdc80 00000000 3fff4274 40216dcb
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld`

marvinroger, 2017-06-23:
Please decode the exception: https://github.com/me-no-dev/EspExceptionDecoder

Can you please post a simple sketch to reproduce? Also, please tell what is the topic that makes it crash.

gorec2005, 2017-06-23:
sorry... it's my mistake in subscribe... you lib awesome!!!!! thank you

closed 2017-06-23 by gorec2005