marvinroger/async-mqtt-client #52: Error httpupdate

windymonkey, 2017-06-26:
Hi,
I tried using ESP8266httpUpdate OTA with my ESP8266 07 - arduino 1.8.2 - esp8266 2.3.0 core along with async-mqtt-client . It returned this error message:
 HTTP_UPDATE_FAILD Error (-104): Wrong HTTP codeStart
I tried using httpupdate with other libs (not async), it worked well. The include list is below:
```
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <EEPROM.h>
#include <Ticker.h>
#include <ArduinoOTA.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <string.h>
#include <PubSubClient.h>
```
Can some one help me please. Thanks.

marvinroger, 2017-06-26:
What if you disconnect the MQTT client before trying to update?

windymonkey, 2017-06-26:
Thank you, but still fail.

> Disconnected from MQTT.
> HTTP_UPDATE_FAILD Error (-104): Wrong HTTP codeConnecting to MQTT...

allanbartley, 2017-06-29:
I had the same problem; what I ended up doing:

```
void onMqttDisconnectUpdate(AsyncMqttClientDisconnectReason reason) {
  // Handle upgrade here.
  // In my code, I have a state machine which will run the update on it's next cycle when put in the update state.
  stateSwitch( STATE_UPDATE );
  // As an alternative, you could set a global variable to flag to do an update.
  // Next time you run through your loop(), check that variable and run the update if set.
}

// Change onDisconnect handler.
// The one I was using before had reconnect code, so this prevents that from running.
mqttClient.onDisconnect(onMqttDisconnectUpdate);
// Disconnect.
mqttClient.disconnect();
```

windymonkey, 2017-06-29:
Dear allanbartley,
I modified my code along with your advice. Thank you, it work now.

closed 2017-06-29 by windymonkey