marvinroger/async-mqtt-client #53: AsyncMqttClient::setSecure(bool) -- Need help

timpur, 2017-06-28:
Ive tried very thing i can think off.

Currently using:
Arduino: 1.8.3
ESP Arduino: 2.4.0-rc1
AsyncTCP: Latest git pull (27th)
AsyncMQTT: Latest(27th)

Build: Generic ESP8266 Module, 1mb (64kb)

The Sketch im trying to build (test build script):
"
#define ASYNC_TCP_SSL_ENABLED 1

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <Ticker.h>
#include <AsyncMqttClient.h>

AsyncMqttClient mqttClient;

void setup() {
  // put your setup code here, to run once:

  mqttClient.connect();
  mqttClient.setSecure(true);
}

void loop() {
  // put your main code here, to run repeatedly:

}
"

The Error:
(~ = my dir)
"
C:\Users\~\AppData\Local\Temp\arduino_build_20908\sketch\sketch_jun28a.ino.cpp.o:(.text.setup+0x4): undefined reference to `AsyncMqttClient::setSecure(bool)'

C:\Users\~\Local\Temp\arduino_build_20908\sketch\sketch_jun28a.ino.cpp.o: In function `setup':

C:\Users\~\AppData\Local\Temp\arduino_modified_sketch_817953/sketch_jun28a.ino:14: undefined reference to `AsyncMqttClient::setSecure(bool)'

collect2.exe: error: ld returned 1 exit status
"

The build also tells me this:
Using library ESP8266WiFi at version 1.0 in folder: ...
Using library Ticker at version 1.0 in folder: ... 
Using library async-mqtt-client at version 0.8.1 in folder: ...
Using library ESPAsyncTCP at version 1.0.0 in folder: ...

If any one has any idea?
Tim.

marvinroger, 2017-06-28:
I think you can't define `ASYNC_TCP_SSL_ENABLED` in your sketch. You can
either specify the flag in your build script or define the flag in both the
AsyncMqttClient and ESPAsyncTCP libs `h/hpp` files.

timpur, 2017-06-28:
Thank you. You lead me to the problem. In the quest to enable tls in the libs you mentioned, the compiler complained about redefinition in the async tcp config file. There i found:
"
#ifndef ASYNC_TCP_SSL_ENABLED
#define ASYNC_TCP_SSL_ENABLED 0
#endif
"

changed 0 to 1 and everything works.

I tried many locations of define this, and only this one location worked. Weird but moving on and will start trying tls with homie :) 

Thanks.

allanbartley, 2017-06-29:
I didn't have success using a definition in my sketch, but it will work if you set it in build_flags in platformio.ini (only applicable to PlatformIO).
I have read that there is some way to set build flags in your board definition when using the Arduino IDE, but I haven't tried it myself.

timpur, 2017-06-29:
Its okay I found setting the config in the asyncConfig.h in the Async TCP lib works so all g :). Not the best solution but it works :P

timpur, 2017-06-29:
Different note I'm getting exceptions with TLS testing. Can't get it to connect to http://test.mosquitto.org still debugging the issue. If I can't work it out might have to ask for some more help sorry :P.

timpur, 2017-07-07:
Cant seem to get the ssl example working for test.mosquitto.org on port 8883 with no fingerprint?

all i get from debug is (with esp debugging enabled):
"
Connecting to MQTT...
please start sntp first !
Disconnected from MQTT.
"

allanbartley, 2017-07-18:
Hi timpur,

It seems that some server certificate signature algorithms cause a crash:
SHA1 - OK
SHA224 - OK
SHA256 - OK
SHA384 - Fail
SHA512 - Fail
MD5 - OK

test.mosquitto.com:8883 uses SHA512, which is causing the crash.
I had a look around for free brokers but didn't find anything.
I run a local Mosquitto server on my LAN for testing, using a certificate with an SHA1 signature.

timpur, 2017-07-20:
Awesome thanks for that 
My local instancd using 512
I'll down grade it and see how that goes :)

timpur, 2017-08-27:
Did test using a 256 bit certificate, worked like a charm but still testing reliability of connection with homie.

closed 2017-08-27 by timpur