marvinroger/async-mqtt-client #54: connect/disconnect cycle with FullyFeaturedSSL example

chessweb01, 2017-07-08:
I'm trying the FullyFeaturedSSL example. It works with nossl, but when I try the code with ssl I run into the connect/disconnect cycle. The disconnect reason is 0. The broker log (mosquitto 1.4.11) just says "Client esp8266-d99216 has exceeded timeout, disconnecting".

umitech, 2019-06-05:
The same problem here. 

I did some investigation on the source and I guess that the problem comes from the AsyncTCP library or at least the way of using that.

Can some one help about this?

umitech, 2019-06-07:
After some digging in the codes, I found that it works if we send our packets with some delay. 

It seems that it gets disconnected when we send some packets at a small time. I guess that the problem comes from the fact that the encryption in SSL is time consuming and the processor can't handle that much of packets (around 6 packets with average 100 bytes in size (including topic and the payload)).

The workaround for me was to queue the packets and send them one by one with some delay between them. Not a nice approach but at least it works now.

I hope it helps some one who has the same problem and also I hope some one can solve it with a better approach at the deeper levels.

bertmelis, 2021-02-25:
I'm running a ESP8266 with SSL using the develop branch as we speak. Feel free to reopen if the issue persists.

closed 2021-02-25 by bertmelis

jeroenst, 2021-02-25:
Related to this?
https://github.com/marvinroger/async-mqtt-client/issues/107