marvinroger/async-mqtt-client #58: QOS 1 and 2 publish - automatic retry

UHSL-Marcus, 2017-08-14:
It is not made clear in the doc: Will this library automatically retry publishing any unconfirmed QOS 1 or 2 messages, or is it up the implementation to attempt to publish again if no confirmation was received?

marvinroger, 2017-08-14:
[Memory management](https://github.com/marvinroger/async-mqtt-client/blob/master/docs/3.-Memory-management.md)

> If the TCP window was sufficient to send your packet, the publish method will return a packet ID indicating the packet was sent. Otherwise, a 0 will be returned, and it's your responsability to resend the packet with publish.

😉

UHSL-Marcus, 2017-08-14:
So during QoS 2, if it fails at any point during the flow, it will return 0?

marvinroger, 2017-08-14:
`publish` will return 0 if there was a failure while sending the packet at TCP level.
MQTT, as you know, implements another level of ACK in the protocol level. So if you don't receive the `onPublish` callback with your packet ID, it means the packet was not acknowledged, and you indeed have to send it again. The lib won't automatically retry because it would have to store the message, and memory is expensive on the ESP8266.

UHSL-Marcus, 2017-08-14:
Thank you, that is exactly the answer I was after.

closed 2017-08-14 by UHSL-Marcus

hakam80, 2017-08-14:
If I understood it correctly, we can not say that this Lib support QoS 1 and 2, because we have to do it our selves, right?

marvinroger, 2017-08-14:
@hakam80 no, the client supports QoS 1 and QoS 2. The MQTT spec states:

> Message delivery retry
When a Client reconnects with CleanSession set to 0, both the Client and Server MUST re-send any unacknowledged PUBLISH Packets (where QoS > 0) and PUBREL Packets using their original Packet Identifiers [MQTT-4.4.0-1]. This is the only circumstance where a Client or Server is REQUIRED to redeliver messages.

So it's not required for the client to retransmit failed messages (unless reconnection with clean session set to `false`, this is the only case where the lib is not spec compliant: https://github.com/marvinroger/async-mqtt-client/blob/master/docs/4.-Limitations-and-known-issues.md).

So the client is compatible (try, pub and sub QoS 2 messages, you won't see any error, everything will work). But it is the responsability of your application to handle failures / re-sending.