marvinroger/async-mqtt-client #60: Cannot reconnect

Travis346, 2017-08-31:
First off, this library is awesome; thank you for creating it.  It's rock solid except for when my ESP gets disconnected and gets stuck in the reconnect loop.  Here is the log from the fully featured example:

Disconnected from MQTT.
Connecting to MQTT...
Disconnected from MQTT.
Connecting to MQTT...

I can't trigger the disconnection consistently so I'm not sure what's causing it but getting stuck in the reconnect loop is worse.  The Mosca MQTT server manages other clients & messages just fine while the ESP is caught in this loop.  I assume the ESP has WiFi connection since it's in the MQTT reconnect loop.  I'm using Adafruit' ESP8266 Feather HUZZAH.  Any thoughts as to why this might be happening?

thinkwareltd, 2017-09-07:
My code worked before upgrading ESP to v2.4.0-rc1 (https://github.com/esp8266/Arduino/releases/download/2.4.0-rc1/package_esp8266com_index.json). Now it exhibits the above issue.
Tried downgrading back to http://arduino.esp8266.com/stable/package_esp8266com_index.json
but the same disconnect issue appears

tonton81, 2018-01-03:
the handlers are a PITA but its documented regardless, the handlers shouldnt be running code, delays(), anything that takes alot of time, it cripples the ESP. if you temporarily disable the handlers in setup() dont be surprised to see the ESP become stable. for me i set a variable in the handler and next time the loop() is run it sees the variable and runs the necessary code, this is solid.

tonton81, 2018-01-03:
Fully-featured sketch

See examples/FullyFeatured.ino

Very important: As a rule of thumb, never use blocking functions in the callbacks (don't use delay() or yield()). Otherwise, you may very probably experience unexpected behaviors.

You can go to the API reference.

tonton81, 2018-01-03:
calling WiFi.begin in the handler is an example of “unexpected behavious” when the ESP locks up. the reason the author used Ticker.h in his demo for WiFi.begin is because it has enough time to exit the handler scope to run it normally. You dont really need Ticker if you know how to use a bool.

closed 2021-02-24 by bertmelis