marvinroger/async-mqtt-client #62: ESP32 compatibility

eos1d3, 2017-09-17:
Can this library work with ESP32? If not, is there any plan for ESP32 support?

Thanks!

bertmelis, 2017-11-05:
I just managed to get the non SSL example working on my esp32 board.

The only thing I had to change is the way the sketch is notified on the wifi events. And I used my Ticker library for esp32 (although you could very well directly work with FreeRTOS timers).

Gist: [here](https://gist.github.com/bertmelis/2e080edfd182dc87348213eac4cb0ee6)
Dependencies: https://github.com/me-no-dev/AsyncTCP and https://github.com/bertmelis/Ticker-esp32

bertmelis, 2017-11-05:
@marvinroger Maybe creatie a seperate repo die the ESP32 version? Hence you can have a seperate library.json with the correct dependencies.

sidoh, 2017-12-16:
I would imagine this is blocked on https://github.com/me-no-dev/AsyncTCP/issues/8.

It'd be nice for async-mqtt-client to complain loudly when trying to enable SSL on the ESP32 until support is added.

This is what you see currently:

```
.piolibdeps/AsyncMqttClient/src/AsyncMqttClient.hpp:17:23: fatal error: tcp_axtls.h: No such file or directory
```

AKozhuhov, 2019-02-13:
What about this fork, anybody can explain how to properly use it?
https://github.com/fremouw/AsyncTCP/tree/mbed-tls

phamminhkhac, 2020-03-20:
https://github.com/fremouw/AsyncTCP/tree/mbed-tls
seem awsome but don't know how to config to work with

bertmelis, 2021-02-25:
closing, ESP32 is supported.
TLS is discussed in a separate issue.

closed 2021-02-25 by bertmelis