marvinroger/async-mqtt-client #64: Disconnecting from Broken if publishing 3-4 Continous big messages

arihantdaga, 2017-10-10:
I have seen this issue in fullfeatured.ino example - Everything runs fine, But if i send(Publish) 3-4 big messages to the broker as soon as it's connected, It actually disconnects the device and onDisconnect is fired within 30Seconds(as i set KeepAlive 30 Sec.). I am sharing both the code which i modified and the Serial LOg. 

```
void onMqttConnect(bool sessionPresent) {
  Serial.println("Connected to MQTT.");
  Serial.print("Session present: ");
  Serial.println(sessionPresent);
  uint16_t packetIdSub = mqttClient.subscribe("test/lol", 2);
  Serial.print("Subscribing at QoS 2, packetId: ");
  Serial.println(packetIdSub);
  for(int i =0; i<4;i++){
      uint16_t packetIdPub2 = mqttClient.publish(longTopic.c_str(), 2, true, longMessage.c_str());
      Serial.print("Sending Long Message - Publishing at QoS 1, packetId: ");
      Serial.println(packetIdPub2);
      Serial.println(ESP.getFreeHeap());
  }
  mqttClient.publish("test/lol", 0, true, "test 1");
  Serial.println("Publishing at QoS 0");
  uint16_t packetIdPub1 = mqttClient.publish("test/lol", 1, true, "test 2");
  Serial.print("Publishing at QoS 1, packetId: ");
  Serial.println(packetIdPub1);
  uint16_t packetIdPub2 = mqttClient.publish("test/lol", 2, true, "test 3");
  Serial.print("Publishing at QoS 2, packetId: ");
  Serial.println(packetIdPub2);

}
void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
  Serial.println("Disconnected from MQTT.");
  Serial.print("Disconnect Reason : ");
  Serial.println(static_cast<uint8_t>(reason));

  if (WiFi.isConnected()) {
    mqttReconnectTimer.once(2, connectToMqtt);
  }
}

void setup(){
    mqttClient.setKeepAlive(30).setCleanSession(false);
}
```

The Serial Output is - 

```
Connecting to Wi-Fi...
Connected to Wi-Fi.
Connecting to MQTT...
Connected to MQTT.
Session present: 1
Subscribing at QoS 2, packetId: 1
Sending Long Message - Publishing at QoS 1, packetId: 2
39792
Sending Long Message - Publishing at QoS 1, packetId: 3
39792
Sending Long Message - Publishing at QoS 1, packetId: 4
38272
Sending Long Message - Publishing at QoS 1, packetId: 5
38272
Publishing at QoS 0
Publishing at QoS 1, packetId: 6
Publishing at QoS 2, packetId: 7
Disconnected from MQTT.
Disconnect Reason : 0
Connecting to MQTT...
Connected to MQTT.
Session present: 1
Subscribing at QoS 2, packetId: 1
Sending Long Message - Publishing at QoS 1, packetId: 2
34776
Sending Long Message - Publishing at QoS 1, packetId: 3
34776
Sending Long Message - Publishing at QoS 1, packetId: 4
33256
Sending Long Message - Publishing at QoS 1, packetId: 5
33256
Publishing at QoS 0
Publishing at QoS 1, packetId: 6
Publishing at QoS 2, packetId: 7
```

I used HiveMq Broker and before that i used Emqttd Broker. Its happening same with both the brokers.

arihantdaga, 2017-10-16:
@marvinroger ... Anything on this one ?

marvinroger, 2017-10-16:
This is not a problem with the available heap, but with the TCP buffer being filled. But the code should be protected against that, in theory... I'll check whenever I can.

bcatalin, 2017-10-16:
How big are the messages?

arihantdaga, 2017-10-16:
```
String longTopic = "/KU/5ccf7ff8793d/58995fb233edd039b1eedf69/ping/get";

String longMessage = "andnansdasndasdnkj12jb312h31!@#$%^&*(dgfchvbjka.smdlkahsdc ashjkdnabvsfdcvbkaln;lskj1b23242356472346374823764732943267432890489726453789248728645327904873264567230943672456387294736287423mbncvbhjklbvg;ljnjhasbcnlaksjbdsjcnksjdbcjhvbkj1hbg23253726138hkjdshvcdbkcjlfnckbhjghdfbjlkdsbcvhgdsjklcndsbhjgcvbsnlksndbgcdsvhbklcbvghsdcbnlskagtrtyuiopoiuytreqwertyuiopoiuytfdsdfghjkl;lkjhgfdsxcvbnm,.,mnbvcxzxcvbnm,.kjhgfdfghjklkjhgfdfghjk";
```

This is what i used.

arihantdaga, 2017-10-20:
@bcatalin  Any Findings ??

arihantdaga, 2017-10-20:
I wrote few debug messages in **asyncmqttclient.cpp Publich funciton**. 
After that debug log is something like this.. 

```
Connected to Wi-Fi.
Connecting to MQTT...
Connected to MQTT.
Session present: 0
Subscribing at QoS 2, packetId: 1
Connected: 1 
Payload Length: 286, Topic Length: 50 
Remaining Length : 2 
Need Space : 343 
Client Space : 2905
Sending Long Message - Publishing at QoS 1, packetId: 2
39464
Connected: 1 
Payload Length: 286, Topic Length: 50 
Remaining Length : 2 
Need Space : 343 
Client Space : 2562
Sending Long Message - Publishing at QoS 1, packetId: 3
39464
Connected: 1 
Payload Length: 286, Topic Length: 50 
Remaining Length : 2 
Need Space : 343 
Client Space : 2219
Sending Long Message - Publishing at QoS 1, packetId: 4
39464
Connected: 1 
Payload Length: 286, Topic Length: 50 
Remaining Length : 2 
Need Space : 343 
Client Space : 1876
Sending Long Message - Publishing at QoS 1, packetId: 5
39464
Connected: 1 
Payload Length: 6, Topic Length: 8 
Remaining Length : 1 
Need Space : 18 
Client Space : 1533Publishing at QoS 0
Connected: 1 
Payload Length: 6, Topic Length: 8 
Remaining Length : 1 
Need Space : 20 
Client Space : 1515Publishing at QoS 1, packetId: 6
Connected: 1 
Payload Length: 6, Topic Length: 8 
Remaining Length : 1 
Need Space : 20 
Client Space : 1495Publishing at QoS 2, packetId: 7
Disconnected from MQTT.
Disconnect Reason : 0
Connecting to MQTT...
Connected to MQTT.
Session present: 0
Subscribing at QoS 2, packetId: 1
Connected: 1 
Payload Length: 286, Topic Length: 50 
Remaining Length : 2 
Need Space : 343 
Client Space : 2905
Sending Long Message - Publishing at QoS 1, packetId: 2
36136
Connected: 1 
Payload Length: 286, Topic Length: 50 
Remaining Length : 2 
Need Space : 343 
Client Space : 2562
Sending Long Message - Publishing at QoS 1, packetId: 3
36136
Connected: 1 
Payload Length: 286, Topic Length: 50 
Remaining Length : 2 
Need Space : 343 
Client Space : 2219
Sending Long Message - Publishing at QoS 1, packetId: 4
36136
Connected: 1 
Payload Length: 286, Topic Length: 50 
Remaining Length : 2 
Need Space : 343 
Client Space : 1876
Sending Long Message - Publishing at QoS 1, packetId: 5
36136
Connected: 1 
Payload Length: 6, Topic Length: 8 
Remaining Length : 1 
Need Space : 18 
Client Space : 1533
Publishing at QoS 0
Connected: 1 
Payload Length: 6, Topic Length: 8 
Remaining Length : 1 
Need Space : 20 
Client Space : 1515Publishing at QoS 1, packetId: 6
Connected: 1 
Payload Length: 6, Topic Length: 8 
Remaining Length : 1 
Need Space : 20 
Client Space : 1495Publishing at QoS 2, packetId: 7
```

Note these - 
**Payload Length: 286, Topic Length: 50 
Remaining Length : 2 
Need Space : 343 
Client Space : 2905**
**Payload Length: 6, Topic Length: 8 
Remaining Length : 1 
Need Space : 18 
Client Space : 1533**

I understand that client space is decreasing but its still a lot .. isn't it ?

simoncheen, 2017-11-07:
I have the same problem , if the message larger then 128, the 8266 will go reboot

arihantdaga, 2017-11-16:
@marvinroger Any updates ? Please help...

marvinroger, 2017-11-16:
Can someone provide a minimum code example to reproduce?

arihantdaga, 2017-11-16:
@marvinroger Thank you...
I gave complete code above.. But you can reproduce it using fullFeatured.ino in the library's example code only... just by replacing messages and topic with these.. 
String longTopic = "/KU/5ccf7ff8793d/58995fb233edd039b1eedf69/ping/get";

```
String longTopic = "/KU/5ccf7ff8793d/58995fb233edd039b1eedf69/ping/get";

String longMessage = "andnansdasndasdnkj12jb312h31!@#$%^&*(dgfchvbjka.smdlkahsdc ashjkdnabvsfdcvbkaln;lskj1b23242356472346374823764732943267432890489726453789248728645327904873264567230943672456387294736287423mbncvbhjklbvg;ljnjhasbcnlaksjbdsjcnksjdbcjhvbkj1hbg23253726138hkjdshvcdbkcjlfnckbhjghdfbjlkdsbcvhgdsjklcndsbhjgcvbsnlksndbgcdsvhbklcbvghsdcbnlskagtrtyuiopoiuytreqwertyuiopoiuytfdsdfghjkl;lkjhgfdsxcvbnm,.,mnbvcxzxcvbnm,.kjhgfdfghjklkjhgfdfghjk";
```

VijayKrishna91, 2017-11-27:
I am also facing similar issue... Disconnecting from broker.. Did anyone find any solution ?

saikatharryc, 2017-12-05:
Did anyone find a solution ??

romansavrulin, 2017-12-15:
Checked that and got the same issue. @marvinroger , Can you give us any points about where to dig to solve this issue?

tarzan115, 2017-12-23:
I got the same issue with a large message, about ~2kb.
how can I split that message and send one by one?

tonton81, 2018-01-01:
mqtt can send up to 260MB payload, but it all depends on the library to support breaking up the packets and reassembling them in callbacks

arihantdaga, 2018-03-02:
@tonton81 Is anyone working on fixing this ??

tonton81, 2018-03-02:
i dont know, havnt seen marvin active in awhile
i ported his library and ESP core to teensy so i can get async mqtt on teensy. the project is a beast that offers wifi stack and core library methods to the host, but havnt had any problems with marvins library, i just never sent huge payloads, im busy doing different things now

timpur, 2018-03-02:
I'd love to have a look into this, but have like no time ATM ...

9a4gl, 2018-03-07:
I was publishing touch events on my screen and if I was touching too fast I got disconnected from server and with Wireshark I saw malformed packet was sent from my ESP32 board, then mosquitto disconnects. Then I changed qos from 2 to 1 and now it works fine, try that.

VijayKrishna91, 2018-03-28:
@9a4gl This wont solve the problem in every case. I am facing it with even QOS 0

fejess, 2018-10-01:
` mqttClient.publish(topic, 1, false, payload);`

QOS 1 solved in my case

9a4gl, 2019-03-21:
I have created pull request that should fix this problem, it is at https://github.com/marvinroger/async-mqtt-client/pull/138
Fix is also on my forked repo https://github.com/9a4gl/async-mqtt-client.git

luebbe, 2021-02-10:
Cannot reproduce anymore with develop

closed 2021-02-10 by luebbe