marvinroger/async-mqtt-client #65: Optimizer feature mqttSend

levipereira, 2017-10-11:
First of all, many thanks for sharing it.
This is not a issue but a improvement.  (sorry for post here, I didn't find the right place).
Suggest add a feature to optimize mqttSend( <topic>, <payload> , <optimize(false default, true)> ). 

This feature enabled avoid resend same payload every loop.
Mqtt optimization skips the payload, if the current payload is same as the last one wich already was sent.
Also is recommended optimization force QoS level 1.

This feature will bring great benefits since we use devices with few hardware resources.

Actually I'm using 15 different TOPIC,  I had to use this:
```
if ( current_payload != last_send_payload){
   mqttSend(<topic>,  current_payload );
    last_send_payload = last_send_payload;
}
```

Thank you again.

tonton81, 2018-01-01:
this would require storing the topic in the heap for long periods of times for every topic that uses it. if you plan to send 512bytes of payloads then you must have a sufficient buffer as well in the library that takes that chunk of ram that your MCU will never be able to use for your own code.

tonton81, 2018-01-01:
plus the time alloted for array comparing on the mcu side before transmitting.

RudyFiero, 2018-01-01:
I think it would be better if you had your logic before generating a payload.

tonton81, 2018-01-01:
that makes more sense, at least the user arranges his own static buffer depending on his needs

bertmelis, 2021-02-10:
This belongs in user code instead of library code.

closed 2021-02-10 by bertmelis