marvinroger/async-mqtt-client #66: Is it safe to tcp_write without the TCP_WRITE_FLAG_COPY?

BrunoMPires, 2017-10-16:
Is there a reason fot the TCP_WRITE_FLAG_COPY not being set for the calls to _client.add issued around this point [here](https://github.com/marvinroger/async-mqtt-client/blob/ddbf4d13287f72bbd281b463f5f032cfe035869f/src/AsyncMqttClient.cpp#L833)? Wouldn't it be safer to guarantee memory allocations on the lower levels?

camilorodegheri, 2017-10-20:
@marvinroger do you know about this?

camilorodegheri, 2017-10-30:
@me-no-dev can you help?

me-no-dev, 2017-10-30:
no need for copy :) it would just cause more memory allocation. I've tested it and made no difference in terms of reliability for sending

marvinroger, 2017-10-30:
I was about to say "the library is made to be as lightweight as possible", but if moreover it does not make any difference, there's not much to say. 😉

closed 2017-11-02 by marvinroger

mknjc, 2020-01-22:
This is very dangerous and wrong.
Lwip need to have all non copied data accessable as long as the corresponding tcp ack isn't received.
This is can't be assured with data on stack and might be part of a lot of bugreports.

Tests may lead to assume it's safe because lwip does sometimes copies the data even without the flag and if it doesn't has to resend the packet it doesn't need to access the pointer. Additionally the data referenced might still be the same because the stack didn't grow again but this is nothing anyone should relay onto.

Also referencing small pieces of memory hurts lwip performance and memory because the PBUF_ROM overhead is rather high.

JonasWtt, 2022-08-10:
Where does tcp_write copy the data to?
I am currently having a problem, when I set tcp_wnd to big, 
the tcp_write only returns mem_err and i can't find the
location where i dont have enough memory!

bertmelis, 2022-08-10:
Where do you set tcp_wnd to big? Asking because the Arduino framework uses pre-compiled nonos and many of the settings are fixed.

Now, fiddling with LWIP isn't exactly part of this library but I'm happy to help where I can.