marvinroger/async-mqtt-client #69: Full topic string when processing message

ikin77, 2017-11-22:
Hi. Let's assume you get a subscription to whathever/request/+. Let's assume someone send a "whathever/request/12345" topic. The client receives the topic, but in the callback function the "topic" string holds the value "whathever/request/" instead of "whathever/request/12345".

Is there a way to get the full topic string?

Thanks in advance and best regards.

marvinroger, 2017-12-01:
Are you mutating the topic char array somewhere? Or using homie-esp8266?

ikin77, 2017-12-15:
Hi. I'm not mutating the topic string, just writting its content to the serial output. I don't know what's "homie-esp8266", so, I guess I'm not using it.

Best regards

bertmelis, 2021-02-10:
I cannot reproduce, feel free to reopen if the issue persists.

closed 2021-02-10 by bertmelis