marvinroger/async-mqtt-client #72: 压力测试,好多问题.

wlzzqkrs, 2017-12-01:
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400d2eff  PS      : 0x00060d30  A0      : 0x800d2f39  A1      : 0x3ffdcb60  
A2      : 0x3ffc317c  A3      : 0x00000000  A4      : 0x3ffdf2be  A5      : 0x000002e2  
A6      : 0x3ffc327c  A7      : 0x00000165  A8      : 0x00000003  A9      : 0x00000079  
A10     : 0x00000000  A11     : 0x00000080  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x0000002a  A15     : 0x3ffd1d00  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d2eff:0x3ffdcb60 0x400d2f36:0x3ffdcbc0 0x400d4846:0x3ffdcbe0 0x400d4a01:0x3ffdcc30

marvinroger, 2017-12-01:
What's that? It does not look like an ESP8266/ESP32 stackstrace, right?

marvinroger, 2017-12-01:
Please speak english, so everyone can understand.

closed 2017-12-08 by marvinroger