marvinroger/async-mqtt-client #73: Garbled data in buffer onMqttMessage

arjhun, 2017-12-20:
First of all, Thanks for your library it is great. I finally got wifi mqtt a sensor and updating an oled screen working at high speed! 

One thing, When I receive mqtt messages from my broker all at once (in this case 4 switches) my payload buffer contains some extra characters...
```
ON
ON0"
ON0"
ON
```
and
```
OFF
OFF0#
OFF0#
OFF
```

I guess this is expected behaviour but I thought I would verify if this is the case... I 
Before I handled it like this, but I believe this copies the entire buffer!?

```String s_payload = String(payload).subString(index,len);```

Now I handle this the following way:
```
String s_payload;

for(int i = index; i < len; i++){
  s_payload+=payload[i];
}
```
I hope I'm doing this the right way! Would it be an idea to add an example to the library that actually processes the incoming data? If not, I understand! the library is very low level and that's good!

tonton81, 2017-12-31:
yeah but you should use character arrays and not the String class unless you like to make swiss cheeze out of your heap. 
you could do something like this for the char array:
char _copied[len+1]; // size for temporary array with an extra spot for a NULL (if needed)
strlcpy(_copied, payload, len); // copy the array, up to len limit max 
_copied[len] = '\0'; // null terminate final char of the new array

arjhun, 2018-01-03:
Thanks so much for pointing me into the right direction! I'll close this "issue"

closed 2018-01-03 by arjhun