marvinroger/async-mqtt-client #74: how can I publish with large message, about 3kb

tarzan115, 2017-12-23:
first of all, thank you for your library.
I have a message with 3kb but I can't send in normal. I had search but I can't find anywhere to do it.

thank you so much.

toomasz, 2018-02-03:
I've got similar problem. Problem seems to be caused by tcp client not having enough buffer space:
https://github.com/marvinroger/async-mqtt-client/blob/0605524875dd4bf3004282812a203f8189d83005/src/AsyncMqttClient.cpp#L827

romansavrulin, 2018-02-03:
@toomasz Thank you for digging that problem and pointing us what's the reason of such behavior!

It seems like AsyncMqttClient tries to publish to the topic using single TCP packet. So, I think we need to split payload across multiple ones.

To solve that, we should
* add internal storage for data, that is passed to publish calls 
* use [`_onAck`](https://github.com/marvinroger/async-mqtt-client/blob/0605524875dd4bf3004282812a203f8189d83005/src/AsyncMqttClient.cpp#L366-L370) internal method to get a new portion of stored data from that storage.

https://github.com/marvinroger/async-mqtt-client/blob/0605524875dd4bf3004282812a203f8189d83005/src/AsyncMqttClient.cpp#L366-L370

tonton81, 2018-02-03:
are you guys trying the latest core in the arduino IDE? try to use the “git version” of esp8266, not the boards manager in the IDE, its not the same core. the IDE comes with a “stable” release while the “git version” has alot of fixes and patches for the problems that are currently in the arduino version..

tonton81, 2018-02-03:
toomasz, those options are available when using the git version of esp8266 :)

tonton81, 2018-02-03:
pubsubclient can “frame” large packets that you’d need to rebuild when receiving, dont know if theres a size limit, but its always updated, maybe you can run both libraries together, use publish with pubsubclient and everything else with asyncmqtt, just to test i guess?

romansavrulin, 2018-02-04:
@tonton81, Thank you for you point, but I use custom arduino workspace with libraries as submodules, pointing to HEAD of master.

If you carefully look through the code, you will mention that reassembly code is only done to packet receiving process, not sending one. But we are trying to send a packet larger than TCP frame can handle. This yet to be implemented in the library.

jeroenst, 2019-01-23:
I adapted this library to use a buffered tcp connection which worked for my problem before, maybe it can also solve your problem. 

https://github.com/marvinroger/async-mqtt-client/issues/121

MikaelHTW, 2019-01-23:
Hi jeroenst, thanks for your fast reply. I'll try it later at home and give you a feedback.

MikaelHTW, 2019-01-23:
Hey, your solution is not working, because i use the asynctcp library for the esp32 instead of the ESPAsyncTCP library for the esp8266.  After the changes I got a " 'AsyncTCPbuffer' does not name a type" error

elloza, 2019-12-16:
Any update on this? it is working?

jeroenst, 2019-12-16:
I don't think it's a very good idea to use a big payload for mqtt. It's better to use multiple topics with smaller chunks of data. In my devices each variable has it's own topic. A big advantage of this approach is that another device can select only the topics that are needed and doesn't receive a lot of unnecessary data.

joelgamal, 2020-02-16:
I am facing the same issue, how to solve it please?

bertmelis, 2020-02-16:
How easy it is to implement depends on the size of your message. Does it fit in memory? Or are you even thinking bigger?

pierreh, 2020-03-11:
I would like to have this feature as well. I am currently probing the ESP32-cam via http using node-red to get an image snapshot and to publish it to MQTT, however this could be done straight from the ESP32 and then I don't need to run a http server. The image buffer is already contained in memory, so there is no need for additional memory. The http library doesn't have an issue sending the whole image in the response. 
Similarly I would like to be able to receive firmware over MQTT to simplify OTA deployment. In this case we should be able to receive the message in chunks and write the data to the OTA partition.

bertmelis, 2020-03-11:
Receiving in chunks is already there. OTA over MQTT is rather straightforward to implement.

bertmelis, 2021-02-10:
Please use the `develop` branch. Sending large payloads is supported in this branch. for now, the limit is the available heap memory.

I'm closing this issue for now. Should you encounter any problems with the `develop` branch, please reopen or open a new issue.

closed 2021-02-10 by bertmelis