marvinroger/async-mqtt-client #75: Got exception 28 when TCP disconnected

tarzan115, 2017-12-28:
some time when I got disconnected TCP I got `exception 28` with error

```
Disconnected from MQTT.
TCP disconnected

Fatal exception 28(LoadProhibitedCause):
epc1=0x4025169f, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000002c, depc=0x00000000

Exception (28):
epc1=0x4025169f epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000002c depc=0x00000000

ctx: sys 
sp: 3ffffc90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe30:  3ffffe60 00000002 00000000 3ffef980  
3ffffe40:  3ffffe90 3ffffed0 3fff0de8 3fff515c  
3ffffe50:  3fff0de8 00000000 3fff0de8 4020f538  
3ffffe60:  3fff00e0 3fff53d4 3fff4ea4 40225e8c  
3ffffe70:  40235459 00000000 00000000 4020f5c5  
3ffffe80:  fffffffd 3fff53d4 00000000 402109eb  
3ffffe90:  4000050c 3fff60fc 00000000 3fff515c  
3ffffea0:  3fff5fbc 00000000 00000000 40210a28  
3ffffeb0:  4021a6eb 3fff19e0 4021a75c 4021190a  
3ffffec0:  3fff53d4 3fff515c 00000000 40211e68  
3ffffed0:  3fff0de8 3fffc6fc 00000001 3fff19e0  
3ffffee0:  00000000 3fffdad0 3fff319c 40211e7c  
3ffffef0:  00000002 3fff5fbc 3fff515c 40251a49  
3fffff00:  00000000 3ffec4ea 3ffec502 40251bac  
3fffff10:  3fff18dc 000000f0 000000f0 00000000  
3fffff20:  3fff2f28 3fff2f24 0320880e 40251c24  
3fffff30:  40249224 3ffeeef8 3fff60e4 40253048  
3fffff40:  401049e2 0320a68b 3ffeeef8 402531dd  
3fffff50:  40253040 00000000 0320a68b 00000000  
3fffff60:  40257ec8 3fff0788 3fff07b0 60000600  
3fffff70:  0ccec568 3fff07b0 3fff0788 40257ed5  
3fffff80:  40257f1a 3fffdab0 00000000 3fffdcb0  
3fffff90:  3fff07c8 3fffdad0 3fff19d8 4021a6eb  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

after decoded
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 29 results
0x4025169f: tcp_recved at core/tcp.c line 605
0x4025169f: tcp_recved at core/tcp.c line 605
0x4020f538: AsyncMqttClient::_sendDisconnect() at C:\Users\doanh\Documents\Arduino\libraries\async-mqtt-client-0.8.1\src/AsyncMqttClient.cpp line 92
0x40225e8c: ssl_read at ssl/tls1.c line 2007
0x40235459: ieee80211_parse_beacon at ?? line ?
0x4020f5c5: AsyncMqttClient::disconnect(bool) at C:\Users\doanh\Documents\Arduino\libraries\async-mqtt-client-0.8.1\src/AsyncMqttClient.cpp line 92
0x402109eb: AsyncMqttClient::_onPoll(AsyncClient*) at C:\Users\doanh\Documents\Arduino\libraries\async-mqtt-client-0.8.1\src/AsyncMqttClient.cpp line 92
0x40210a28: _M_invoke at C:\Users\doanh\Documents\Arduino\libraries\async-mqtt-client-0.8.1\src/AsyncMqttClient.cpp line 92
0x4021a6eb: loop_task at C:\Users\doanh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_main.cpp line 132
0x4021a75c: loop_wrapper at C:\Users\doanh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_main.cpp line 113
0x4021190a: std::function ::operator()(void*, AsyncClient*) const at C:\Users\doanh\Documents\Arduino\libraries\ESPAsyncTCP-master\src/ESPAsyncTCP.cpp line 659
0x40211e68: AsyncClient::_poll(tcp_pcb*) at C:\Users\doanh\Documents\Arduino\libraries\ESPAsyncTCP-master\src/ESPAsyncTCP.cpp line 659
0x40211e7c: AsyncClient::_s_poll(void*, tcp_pcb*) at C:\Users\doanh\Documents\Arduino\libraries\ESPAsyncTCP-master\src/ESPAsyncTCP.cpp line 659
0x40251a49: tcp_slowtmr at core/tcp.c line 969 (discriminator 1)
0x40251bac: tcp_fasttmr at core/tcp.c line 1032 (discriminator 1)
0x40251c24: tcp_tmr at core/tcp.c line 127
0x40249224: ppRecycleRxPkt at ?? line ?
0x40253048: tcpip_tcp_timer at core/timers.c line 87
0x401049e2: wdt_feed at ?? line ?
0x402531dd: sys_check_timeouts at core/timers.c line 420
0x40253040: tcpip_tcp_timer at core/timers.c line 81
0x40257ec8: ets_timer_handler_isr at ?? line ?
0x40257ed5: ets_timer_handler_isr at ?? line ?
0x40257f1a: ets_timer_handler_isr at ?? line ?
0x4021a6eb: loop_task at C:\Users\doanh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_main.cpp line 132
```

tonton81, 2018-01-04:
lots of exception 28 in 2.4 milestone of esp8266, is that what your on? if so, the issue is there and not here

tonton81, 2018-01-04:
just saw your log :) roll back to 2.2/2.3 and try again.

tarzan115, 2018-01-05:
I fix it with this PR: https://github.com/me-no-dev/ESPAsyncTCP/pull/75.

luebbe, 2021-02-10:
Closing, since the PR is merged in ESPAsyncTCP

closed 2021-02-10 by luebbe