marvinroger/async-mqtt-client #79: Consecutive publishes. only first two get to broker

gjt211, 2018-01-28:
Using the most recent Async libraries from about 1 week ago, with the Arduino IDE on OSX, and ESP8266.

In my `onMqttConnect` callback, I was publishing 4 values to the broker (mosquitto) and this was working well until I updated my libs about 1 week ago.
Now only the first one or two (of the four) actually get sent.

I created a `Ticker` that runs every second, and now only set a flag in the `onMqttConnect` callback. In the Ticker, I check if a flag has been set, and if so, I clear that flag, perform the publish and return. This way only one item can get published each time the Ticker is called.

I have tried reducing the Ticker interval and start to get problems with anything less than about 0.3 seconds.

It seems if I publish consecutively to several topics, regardless of where I do that (i.e. NOT in the callback) then only some of them actually get to the broker.

Has any changes been made in the last few months that may cause anything like this to happen?
My free heap varies between about 20000 and 23000 bytes.

9a4gl, 2018-03-07:
I was publishing touch events on my screen and if I was touching too fast I got disconnected from server and with wireshark I saw malformed packet was sent from my ESP32 board. Then I changed qos from 2 to 1 and now it works fine, try that.

romansavrulin, 2018-03-08:
@9a4gl @gjt211 take a look at [that issue](https://github.com/marvinroger/async-mqtt-client/issues/92#issuecomment-371503078)

9a4gl, 2018-03-08:
I haven't monitored free heap, but I do have like 114kB on my ESP32 while idle. I can make it send malformed packet in like 10 fast touches, so I doubt you need 10kB per message. Also note that in my case server closes connection because it receives malformed mqtt message. Wireshark shows it is not complete. And it happen only with qos=2, qos=1, no way, stable as a rock.

romansavrulin, 2018-03-08:
@9a4gl 

This is not about free heap. this is about number of pcbs available in lwip stack. Just add printf here and check if that happens.

```cpp
if (_client.space() < neededSpace) {
    _connectPacketNotEnoughSpace = true;
   Serial.println(F("No free PCBs"));
    _client.close(true);
    return;
  }
```

closed 2018-12-18 by gjt211

9a4gl, 2019-03-20:
I think I have finally found the problem, and seems the problem is related to ESP32 boards due to 2 cores, it happen if you send publish from one core and other core is sending ack. Temporary I have fixed publish and sending ack and I couldn't reproduce the problem in 24h. Previously I got disconnected from server once per hour. Have to find some time to fix other messages and create an PR.

9a4gl, 2019-03-21:
I have created pull request that should fix this problem, it is at https://github.com/marvinroger/async-mqtt-client/pull/138
Fix is also on my forked repo https://github.com/9a4gl/async-mqtt-client.git