marvinroger/async-mqtt-client #80: onMqttPublish callback never gets called

gjt211, 2018-01-28:
I am not sure if this is a problem or not, but my ESP8266 callback for `onMqttPublish` never gets called.

I am wondering if this is because of using QOS0?

My callback is as follows;
```
void onMqttPublish(uint16_t packetId) {
  Serial.print("[MQTT] {PUB} ACK ");
  Serial.println(packetId);
}
```

And in my `setup()` function, I have the following line;
```
mqttClient.onPublish(onMqttPublish);
```

Any help is appreciated, thanks.

tonton81, 2018-01-28:
it works for me, make sure you have the library updated and latest GIT version of esp8266, not the stock one in the IDE

gjt211, 2018-01-28:
Thanks @tonton81, I have been using the library for over 6 months, and have never seen this callback run.
I recently (9th Jan) updated all my libraries (core & all async) to the current versions.

I will try a simple/example sketch now and report back.

tonton81, 2018-01-28:
i must admit you mentioned qos0, i only use qos2 in my project. youll need to try to see if it is a qos level thing

gjt211, 2018-01-28:
Ok, I have just used the example `FullyFeatured-ESP8266` and added my stuff for my secure broker.
From what I can tell with my tests, my results are;
```
QOS=0: The onMqttPublish callback never gets called.
QOS>0: The onMqttPublish callback does get called.
```

I tried with the retain flag true or false and it made no difference.

I am now wondering if this is expected behaviour or not?

gjt211, 2018-01-28:
Further reading on the MQTT protocol shows that QOS=0 messages never get acknowledged.

closed 2018-01-28 by gjt211