marvinroger/async-mqtt-client #81: Can i use onewire lib?

kiralikbeyin, 2018-02-02:
Because it uses delays

````
	
do {
		if (--retries == 0) return 0;
		delayMicroseconds(2);
	} while ( !DIRECT_READ(reg, mask));

	noInterrupts();
	DIRECT_WRITE_LOW(reg, mask);
	DIRECT_MODE_OUTPUT(reg, mask);	// drive output low
	interrupts();
	delayMicroseconds(480);
	noInterrupts();
	DIRECT_MODE_INPUT(reg, mask);	// allow it to float
	delayMicroseconds(70);
	r = !DIRECT_READ(reg, mask);
	interrupts();
	delayMicroseconds(410);
	return r;
}
````

tonton81, 2018-02-02:
why not? shouldnt be an issue, delay() & yield() both satisfy the watchdog and wifi stack...

kiralikbeyin, 2018-02-02:
I have 7 temp sensors.

````
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
Decoding 58 results
0x401058d4: ets_timer_setfn at ?? line ?
0x40105b00: ets_timer_arm_new at ?? line ?
0x40224264: OneWire::write_bit(unsigned char) at /Users/zx/Documents/Arduino/libraries/OneWire/OneWire.cpp line 316
0x402012d8: delay at /Users/zx/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring.c line 45
0x4022431d: OneWire::write(unsigned char, unsigned char) at /Users/zx/Documents/Arduino/libraries/OneWire/OneWire.cpp line 316
0x402248bf: DallasTemperature::blockTillConversionComplete(unsigned char, unsigned char const*) at /Users/zx/Documents/Arduino/libraries/DallasTemperature/DallasTemperature.cpp line 50
0x402248fe: DallasTemperature::requestTemperatures() at /Users/zx/Documents/Arduino/libraries/DallasTemperature/DallasTemperature.cpp line 50
````

kiralikbeyin, 2018-02-02:
First i was getting temperatures then publish which makes trouble.

Now i am first publishing temp data then i get temperatures every minutes.
This is my temporary solution...

Is there any correct way with delaying libs like onewire?

tarzan115, 2018-02-03:
yes, you can.
you have to use `DallasTemperature` by `async` way. set method `setWaitForConversion` to `false`. before call `requestTemperatures()`.
and check the temperature by call `isConversionComplete` in a callback.
sorry for my bad English.
If you can't understand. maybe some small of your code will better.

romansavrulin, 2018-02-05:
@tarzan115 don't worry. Good and readable explanation.

closed 2021-01-27 by bertmelis