marvinroger/async-mqtt-client #85: How to connect ESP8266

Mark-81, 2018-02-18:
Thanks for your work! I have a simple question: how to connect a ESP8266 module to the Arduino? I.e. I have an Olimex one. I need to connect it to the main Arduino UART? I don't see any configuration in the code.

tonton81, 2018-02-18:
why you need to connect it to arduino? you want the arduino as a slave?

Mark-81, 2018-02-18:
Because right now it's the only way I know... I always used an AVR as a "master" that sends AT commands to the ESP8266 module. Without this connection how I can use the ESP8266 module?

tonton81, 2018-02-18:
you program arduino ONTO the esp8266 same as arduino

Mark-81, 2018-02-18:
Got it, thanks! Last question - I apologize. That means I can use any module if I would able to replace the firmware or just place the ESP8266 on my own PCB, right?

tonton81, 2018-02-18:
module i presume you mean library? yeah alot of arduino libraries can run on esp8266 just like arduino does, the arduino IDE flashes the ESP with your sketch and itll work just like an arduino after

Mark-81, 2018-02-18:
No, I'm talking about physical modules, like this: https://www.olimex.com/Products/IoT/MOD-WIFI-ESP8266-DEV/open-source-hardware.

I designed a complex board with an xmega that does a lot of stuff. Then I need to provide WiFi connection. Right now I have the module above connected using the UART and I send AT commands to connect to an AP, open a socket and send some data.

To use this library instead I need to burn a custom firmware on the Olimex module and adapt the example code in order to receive data (i.e. sensors) from the xmega and publish them to the broker.

tonton81, 2018-02-18:
the olimex is nothing but an esp8266 module flashed with AT command set. you could drop all your xmega code on the esp8266 (olimex) and run it without the xmega, if you have other connections on the xmega you can bring them over but be aware some ESP pins are special and can’t be used without caution

tonton81, 2018-02-18:
the ESP also has more ram flash and storage and faster CPU than your xmega, which the AT command set doesnt put to good use :)