marvinroger/async-mqtt-client #88: On publish callback is not called for qos 0 messages

romansavrulin, 2018-02-27:
It should be done either at the end of publish function, or in _onAck for sent tcp packet with publish message

timpur, 2018-02-27:
Good point

Bolukan, 2019-09-23:
After sending some packages (WiFi is expensive for the battery, so I send multiple packages if WiFi connects) my device goes to deep sleep. With qos 0, I cannot check whether all messages left the device. You may guess they don't. So I support this request.
The difference in extra time consumed between qos 0 vs qos 1 I measured to be 10ms+ per package. For the time being my solution is to use qos 1 for the last message, just to be sure the package was fired before deep sleep.

elbueno222, 2021-01-27:
Please correct me if I am wrong but the function "On publish" is called when the ACK is received, what means the server has received the message and sent a confirmation, but QOS 0 doesn´t have ACK so I wouldn´t expect the onMqttPublish function to be call for QOS 0 messages.

I am having exactly the opposite issue now, I am using QOS 1 and some of my messages are not reaching my mqtt server but the function onMqttPublish is called and I don´t understand why.

bertmelis, 2021-02-25:
Behaviour changed in `develop`: all packets are enqueued and handled one after the other. So you can safely publish all your data using qos 0 and immediately disconnect (don't force) afterwards. The disconnect packet will only be sent after the publish packets.

closed 2021-02-25 by bertmelis