marvinroger/async-mqtt-client #92: Subscribe to multiply topics

gamecompiler, 2018-03-06:
Can i subscribe to multiply topics with this lib?
When i subscribe, it seems that i overwrote the old subscription.

timpur, 2018-03-06:
Example ?

gamecompiler, 2018-03-07:
```
Serial.println("Connected to MQTT.");
  Serial.print("Session present: ");
  Serial.println(sessionPresent);

  uint16_t packetIdSub = mqttClient.subscribe("game/info", 2);
  Serial.print("Subscribing game/info at QoS 2, packetId: ");
  Serial.println(packetIdSub);
  
  uint64_t chipid;  
  chipid = ESP.getEfuseMac();

  //https://stackoverflow.com/questions/27392261/arduino-converting-uint64-t-to-string/27393089#27393089
  
  unsigned long long1 = (unsigned long)((chipid & 0xFFFF0000) >> 16 );
  unsigned long long2 = (unsigned long)((chipid & 0x0000FFFF));

  //Subscribe to messages FOR the player
  String chipid_hex = String(long1, HEX) + String(long2, HEX); // six octets
  String player_sub = "game/player/" + chipid_hex;
  char buf[player_sub.length() + 1];
  player_sub.toCharArray(buf, player_sub.length());
  mqttClient.publish(buf, 0, true, "register");

  uint16_t packetIdSub2 = mqttClient.subscribe(buf, 2);
  Serial.print("Subscribing player info at QoS 2, packetId: ");
  Serial.println(packetIdSub2);
  Serial.println(buf);
```

The Client will never receive anything at game/info

timpur, 2018-03-07:
Firstly you can use String.c_str() instead of the buffer

Secondly I'm assuming that you send game/info out side of the device? Could be a QoS thing ?

Also assuming that you've registered the on message callback and stuff.

romansavrulin, 2018-03-07:
@timpur @gamecompiler This library can operate subscriptions and make any other actions step-by-step. So you have to wait for previous action to finish.

This code snippet will help you do the thing.

```cpp

	AsyncMqttClient _mqtt;
        String payloadBuf;

	typedef enum {
		DISC,
		SUB_Auth,
		SUB_Upd,
		SUB_Job,
		ASK_Job,
		RDY
	}connState_t;

	connState_t connState = DISC;

_mqtt.onConnect([this](bool sessionPresent) {
		connState = DISC;
		Serial.println(F("\nMQTT Connected!"));
		auto t = String(F("/info"));
		Serial.print(F("\nSubscribing to "));
		Serial.println(t);
		if (_mqtt.subscribe(t.c_str(), 2))
			connState = SUB_Auth;
		else {
			Serial.print(F("\nDisconnect on "));
			Serial.println(connState);
		}
	}).onSubscribe([this](uint16_t packetId, uint8_t qos) {
		switch (connState) {
		case SUB_Auth:
		{
			auto t = String(F("Auth"));
			Serial.print(F("\nSubscribing to "));
			Serial.println(t);
			if (_mqtt.subscribe(t.c_str(), 2))
				connState = SUB_Upd;
			else {
				Serial.print(F("\nDisconnect on "));
				Serial.println(connState);
			}
		}
		break;
		case SUB_Upd:
		{
			auto t = String(F("update/+"));
			
			Serial.print(F("\nSubscribing to "));
			Serial.println(t);
			if (_mqtt.subscribe(t.c_str(), 2))
				connState = SUB_Job;
			else {
				Serial.print(F("\nDisconnect on "));
				Serial.println(connState);
			}
		}
		break;
		case SUB_Job: {
			connState = ASK_Job;
			Serial.println(F("\nSubscribed all topics!"));
			auto t = String(F("/new"));
			if (_mqtt.publish(t.c_str(), 1, 0)) {
				Serial.print(F("\npub to "));
				Serial.println(t);
			}
			else {
				Serial.print(F("\nDisconnect on "));
				Serial.println(connState);
			}
			break;
		}
		default:
			Serial.println(F("\nUnknown State"));
			_mqtt.disconnect();
			break;
		}
	}).onDisconnect([this](AsyncMqttClientDisconnectReason reason) {
		Serial.println(F("\nMQTT Disconnected!"));
		if (WiFi.isConnected())
			_mqtt.connect(); //restore connection
	}).onMessage([this](char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {
		if (index == 0) {
			payloadBuf = "";
		}

		auto pl = len;
		auto p = payload;
		while (pl--) {
			payloadBuf += *(p++);
		}

		if (index + len == total) {
				Serial.println(F("\nFailed parse JSON from payload: "));
				Serial.println(payloadBuf);
		}

		return;
	}).onPublish([this](uint16_t packetId) {
		switch (connState) {
		case ASK_Job: {
			connState = RDY;
			auto t = String(F("request"));
			if (_mqtt.publish(t.c_str(), 2, 0)) {
				Serial.print(F("\nAsk new job to "));
				Serial.println(t);
			}
			else {
				Serial.print(F("\nDisconnect on "));
				Serial.println(connState);
				_mqtt.disconnect();
			}
			break;
		}
		}
	});
}
```

To simplify state-machine logic, it would be better to implement promises. Maybe take a look into it in near future, because we have a demand on it.

timpur, 2018-03-08:
And not sure about this issue, form memory I can make multiple subs in a row with out waiting for them to resolve and then make pubs

romansavrulin, 2018-03-08:
@timpur in general, if you luck enough not to eat all space in TCP out queue, you can send sequential requests.  And if you eat all space, you will get sudden a disconnect. That's what most of people experiencing here. This obviously should be fixed.

Take a look at implementation.

```cpp
if (_client.space() < neededSpace) {
    _connectPacketNotEnoughSpace = true;
    _client.close(true);
    return;
  }

  _client.add(fixedHeader, 1 + remainingLengthLength);
  _client.add(protocolNameLengthBytes, 2);
  _client.add("MQTT", protocolNameLength);
  _client.add(protocolLevel, 1);
  _client.add(connectFlags, 1);
  _client.add(keepAliveBytes, 2);
  _client.add(clientIdLengthBytes, 2);
  _client.add(_clientId, clientIdLength);
  if (_willTopic != nullptr) {
    _client.add(willTopicLengthBytes, 2);
    _client.add(_willTopic, willTopicLength);

    _client.add(willPayloadLengthBytes, 2);
    if (_willPayload != nullptr) _client.add(_willPayload, willPayloadLength);
  }
  if (_username != nullptr) {
    _client.add(usernameLengthBytes, 2);
    _client.add(_username, usernameLength);
  }
  if (_password != nullptr) {
    _client.add(passwordLengthBytes, 2);
    _client.add(_password, passwordLength);
  }
  _client.send();
```

timpur, 2018-03-08:
I see

arihantdaga, 2018-04-21:
@romansavrulin . I checked in the code and saw that this particular snippet is written only in onConnect Method.

```
if (_client.space() < neededSpace) {
    _connectPacketNotEnoughSpace = true;
    _client.close(true);
    return;
  }
```

But do you think it should be there in` _sendPing() `as well as` publish` methods as well. 
Where right now these function are only returning false but not closing the connection 
`if (_client.space() < neededSpace) return false;`

romansavrulin, 2018-04-21:
@arihantdaga I think it's a bad idea to have disconnect logic when TCP output queue is full elsewhere in the methods which making requests to server. We will be getting unexpected behavior from method then. Say, you call publish, but if you cannot publish due to full queue you do not return an error about it but disconnect instead. That's not what we want. Returning an error is the best we can do in that case.

timpur, 2018-04-22:
Sorry guys, I do plan on fixing these things, but find it hard to find time ATM. PRs are always a help :)

9a4gl, 2019-03-21:
I have created pull request that should fix this problem, it is at https://github.com/marvinroger/async-mqtt-client/pull/138
Fix is also on my forked repo https://github.com/9a4gl/async-mqtt-client.git

closed 2021-02-24 by bertmelis